eclipse-mraa/mraa #841: error reading uart information

kvkdileep, 2017-11-02:
Iam using mraa on raspberry pi 3 which is running raspbian stretch
when i try to list the uart bus. It is throwing error
# mraa-uart list
   0. ---- error reading uart information ----

Can anyone help me resolve the issue ?

Mani-Sadhasivam, 2017-11-02:
Could you please post the relevant libmraa logs? By default, the logs are stored in /var/log/syslog

Mani-Sadhasivam, 2017-11-02:
@kvkdileep I see the issue now! If the UART port you are trying to access is already occupied for console output, you will likely see that error message.

It looks like from the error there is only one UART available in Raspberry pi3 and that is assigned as the console out. So, as a work around you can release the UART console by removing `console=tty***115200` parameter in kernel bootargs, but for that you have to recompile the kernel.

kvkdileep, 2017-11-02:
I removed the console=tty***115200 from /boot/cmdline.txt and rebooted, but still the same. 

In syslog it shows only two lines 

Nov  2 10:25:30 raspberrypi systemd-udevd[750]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Nov  2 10:25:38 raspberrypi kernel: [   64.174233] random: crng init done
Nov  2 10:27:07 raspberrypi libmraa[810]: libmraa version v1.7.0-73-g615e0cd initialised by user 'pi' with EUID 1000
Nov  2 10:27:07 raspberrypi libmraa[810]: libmraa initialised for platform 'Raspberry Pi 3 Model B' of type 5

kvkdileep, 2017-11-14:
Could some one help me out of this issue, desperately in need of help !

Mani-Sadhasivam, 2017-11-14:
I am not familiar with Rpi3 but it looks like there are 2 serial ports and one is used by the console and another one is used for Bluetooth.

So, it would be great if you can disable both and login using the keyboard/monitor then execute `mraa-uart list`.

For disabling Bluetooth, add the following line in `/boot/config.txt`
`dtoverlay=pi3-disable-bt`

For disabling console, remove `console=serial0,115200 `or `console=ttyAMA0,115200` in `/boot/cmdline.txt`.

Now, reboot and let us know.

kvkdileep, 2017-11-14:
I have done the same, but same situation here. Here are the files attached.
[cmdline.txt](https://github.com/intel-iot-devkit/mraa/files/1470350/cmdline.txt)
[config.txt](https://github.com/intel-iot-devkit/mraa/files/1470351/config.txt)

kvkdileep, 2017-11-14:
Iam using raspbian stretch image for pi3, is it a problem ?

Mani-Sadhasivam, 2017-11-14:
Not sure! Could you please remove `dtoverlay=pi3-miniuart-bt` from config.txt and `console=tty1` from cmdline.txt?

kvkdileep, 2017-11-14:
tried the same. Here are the attached files & the result after reboot
[cmdline.txt](https://github.com/intel-iot-devkit/mraa/files/1470405/cmdline.txt)
[config.txt](https://github.com/intel-iot-devkit/mraa/files/1470406/config.txt)

pi@raspberrypi:~ $ mraa-uart list
0. ---- error reading uart information ----
pi@raspberrypi:~ $

Mani-Sadhasivam, 2017-11-14:
@kvkdileep Got the issue! There is no `device_path` registered for UART in the Raspberry pi Source. 

Only way to get over this issue is to add ` b->uart_dev[0].device_path = "/dev/serial0";` after [this](https://github.com/intel-iot-devkit/mraa/blob/master/src/arm/raspberry_pi.c#L836) line in the mraa source and rebuild it.

> Note: I assumed `/dev/serial0` is a symbolic link to correct serial port. Please cross check it.

Mani-Sadhasivam, 2017-11-14:
Let me know if this works for you. I will submit a patch to update the source!

kvkdileep, 2017-11-14:
Holy Moly ! that saved my day.
pi@raspberrypi:~/mraa/mraa/build $ mraa-uart list
0. /dev/serial0 <undefined>         9600 8N1 (none) XONXOFF
pi@raspberrypi:~/mraa/mraa/build $ 
Thanks Mani, it works. You may update the source with the patch

Mani-Sadhasivam, 2017-11-14:
Cool. Fix submitted #842. 

You can close the issue now!

closed 2017-11-14 by kvkdileep

alext-mkrs, 2017-11-14:
...or better yet - you can [include](https://help.github.com/articles/closing-issues-using-keywords/) "Fixes #841" in the commit message and then GitHub will close the issue as soon as the PR is merged (i.e. when it's really fixed).

Mani-Sadhasivam, 2017-11-15:
@alext-mkrs Not aware of this! Will do it for upcoming fixes ;-) Thanks.