eclipse-mraa/mraa #843: What does "low speed" IO mean in the description?

iSsO, 2017-11-28:
I didn't find a description for why mraa is considered low speed. I mean it should be slower compared to something. What is it?

I was wondering, is that because of its implementation or because of how Linux works in general?

Does anyone know how it compares to WiringPi for instance?

arfoll, 2017-11-28:
Low speed I/O refers to SPI, i2c, uart generally. These are slow enough that you can use them from userspace without *too* many issues for general usage. I/O such as PCIe or QPI are I/Os that I would consider 'high speed'. Maybe the description is misleading....

Mraa isn't going to be any slower to wiringPI.

iSsO, 2017-11-28:
Thanks, got it.

The reason I'm asking that is I'm considering using mraa for building a Linux board that can run Arduino code. I know it won't be comparable to a native bare-metal board in terms of latency and determinism, but I need Linux for other stuff.

That's why I'm looking for candidates for GPIO, SPI, I2C and UART access libraries from the user space.

As a main maintainer to mraa, do you think it is a good candidate for my app?

closed 2017-11-28 by iSsO

arfoll, 2017-11-28:
So yes, I think this would work well. FYI the Arduino team is using mraa
below libwiring for a bunch of x86 platforms already. Try Arduino create
with an x86 board and export it, you'll see libmraa in there :)

iSsO, 2017-11-28:
@arfoll That is reassuring, I will check them out, thanks again! :)