eclipse-mraa/mraa #846: Cannot use mraa with snap!byob while rpi.gpio works...

g-vidal, 2017-12-04:
Hi all,
I am using a raspberry Pi with mrra / upm libs with Jupyterhub It works really great ! Some notebooks with examples in french  ;-(  [are available here](https://github.com/g-vidal/CahierDeProgrammes) if of any interest. 
I would like to reuse the work done with a snap!byob interface connected to a python web service running on the Pi . I have found @pbrown66 [work](https://github.com/pbrown66/snap-RPi) that does exactly what I want but unfortunately with rpi.GPIO. I have tried to change rpi.GPIO calls with mraa calls but it does not work : 
```shell
#		GPIO.setmode(GPIO.BCM)
.../...
#			GPIO.setup(self.pin, GPIO.OUT)
#			GPIO.output(self.pin, self.state)

with
			led = mraa.Gpio(self.pin)
			led.dir(mraa.DIR_OUT)
			led.write(self.state)
```
Can anybody push me on the right track telling me what I am missing, or give the solution. So far I have investigated issues linked with pin numbering and access rights to the GPIO (my user belongs to GPIO group who has full access to the GPIO device which enables the original program to lit a led but I get nothing whith my mraa version without finding meaningful traces or errors; if I start the python server as sudo, I get a short faint illumination). I have started a repo [snap-RPi-mraa](https://github.com/g-vidal/snap-RPi-mraa).
Thanks

alext-mkrs, 2017-12-04:
At the first glance, the code should work. I'd look at the user the whole thing runs under. You said you checked that, but I'd simplify and start with the same mraa code (pin setup, write), extracted to a standalone py file. Run it as that webserver user and see if that works. If that's a permissions problem, you should see that in syslog - mraa will complain.

If not that - I'd check pin numbers you get, see if that matches mraa numbering.

If nothing works - check syslog for any mraa traces and maybe do a short video, showing what you do and how it looks like, so that we can see correlated events.

g-vidal, 2017-12-06:
**Progress** but not a solution yet...
I followed your suggestions :  if I test the code in raw python it works properly and if, after this test, I start the service and use snap! surprisingly It works !!!!!
Then if I change the pin I got an error message in `/var/log/syslog` like this (pin number is correct)
```shell
Dec  6 01:17:55 3littlehills libmraa[26942]: gpio19: Failed to open 'value': No such file or directory
Dec  6 01:17:55 3littlehills libmraa[26942]: gpio19: dir: Failed to open 'direction' for writing: No such file or directory
```
If I  stop the server (or not) and declare the pin from another window in raw python the new pin like this : 
```shell
Python 2.7.14 (default, Sep 17 2017, 18:50:44) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> led = mraa.Gpio(35)
>>> 
```
The snap! program that was throwing errors few seconds ago works perfectly :-O . I have been able to reproduce  with all the pins I checked and once the pins have been declared  in raw python everything keeps on working. 
I have tried to start the program as sudo, I have tried to add www-data to the gpio group (that grants full acces to GPIO) without any effect. It looks like the declaration  `led = mraa.Gpio(self.pin)` was not creating the file but if the file exists the server runs properly.

Any idea on the origin of the problem?
Thanks

g-vidal, 2017-12-06:
Oups sorry....
The problem seems not to come from mraa but from the python socketserver... persistence or something in the same vein.
Thanks for the help I will publish the solution on my repo

closed 2017-12-06 by g-vidal

alext-mkrs, 2017-12-06:
No worries, glad you have it working :)

g-vidal, 2017-12-10:
At last not so simple see #802 , _ think it is the same problem for me._