eclipse-mraa/mraa #847: UART failed to setup

Mohamad777, 2017-12-20:
I've installed and build mraa:
https://github.com/intel-iot-devkit/mraa

I've compiled and run uart.c in examples folder. But I get this error message:

    "UART failed to setup"

I've found that it may be because of the code bellow (It means I get the error message "uart: device path undefined" in syslog:

    mraa_uart_init_raw(const char* path)
    {
        mraa_result_t status = MRAA_SUCCESS;
        mraa_uart_context dev = NULL;
    
        if (!path) {
            syslog(LOG_ERR, "uart: device path undefined");
            status = MRAA_ERROR_INVALID_PARAMETER;
            goto init_raw_cleanup;
        }
    ...

here is the source: https://github.com/intel-iot-devkit/mraa/blob/master/src/uart/uart.c

Please note that I've disabled the serial console on my Raspberry Pi 3.

arfoll, 2017-12-20:
So mraa_uart_init() which is what the example call is calling requires a registered and mapped uart port which you likely don't have as it's a serial console by default on the RPI3. Try just doing a mraa_uart_init_raw("/dev/ttyXXX").

Mohamad777, 2017-12-20:
Thanks alot, It solved the problem.

alext-mkrs, 2017-12-21:
That means you need to have that in the board config section of the mraa's source code for your board. See src/arm directory for those, you can see examples inside. An alternative is to use [JSON file](https://github.com/intel-iot-devkit/mraa/blob/master/docs/jsonplatform.md), or just use tha `_raw` init function, as suggested by @arfoll.

Mohamad777, 2017-12-24:
I'm using upm library to connect to fingerprint module(zfm20).
It uses ZFM20(int uart); so I cant modify the source of this library, neither I cant modify the source of mraa library! Is it possible? for example modify upm library to something like this: ZFM20(char ttyXX);
And i didn't understand how to solve mraa_uart_init() problem, would you please explain more?
I'll appreciate any hint.

alext-mkrs, 2018-01-16:
As further discussion is already taking place in a proper repo (upm): https://github.com/intel-iot-devkit/upm/issues/621, I'm closing this one.

closed 2018-01-16 by alext-mkrs