eclipse-mraa/mraa #848: How could I remove mraa and install it again?

Mohamad777, 2017-12-21:
I've changed some files of the library manually, so I've problem with mraa now.

How could I remove mraa and install & build it again?

arfoll, 2017-12-21:
Just run make install again? It'll wipe all the old libs and headers.

Mohamad777, 2017-12-21:
It didn't work.
I've removed /usr/lib/arm-linux-gnueabihf/libmraa.so file and now it is not available even after make install again.
And I get this error in run time:
error while loading shared libraries: libmraa.so.1: cannot open shared object file: No such file or directory

I'm using Ubuntu mate on RPI3.

alext-mkrs, 2017-12-21:
Are you sure `make install` succeeds? If you could post a console log for the whole process, that could be helpful for understanding what's going on. The process is way to simple to break :), so I wonder what could've gone wrong there.

alext-mkrs, 2018-01-16:
@Mohamad777, is this still relevant, do you need any further help?

alext-mkrs, 2018-01-28:
Closing for no response, feel free to comment if help is still needed.

closed 2018-01-28 by alext-mkrs

jinglong485, 2022-02-02:
Hi,

Can I usr make uninstall to remove mraa?