eclipse-mraa/mraa #849: How I can crosscompilecompile new version of mraa for linkit 7688

Ladeia, 2017-12-21:
Hello,

My linkit have one old version of lib mraa 0.8, but I need compile atual version for linkit, but I cant see the document for crosscompile.

Someone can help-me?

alext-mkrs, 2017-12-22:
Generally, as we use cmake, that's no different than cross-compiling any cmake-based project. Specifics depend on your toolchain and environment. Is that just a plain compiler/linker/etc set, or something more complex like Yocto? If the former and you have so called toolchain file, you can feed it into cmake and it should do it automagically. For yocto you can use an existing recipe from meta-oe. Describe what you have (if that's freely downloadable - share a link, I'll take a look) and I may be able to advise more specifically.

alext-mkrs, 2018-01-16:
@Ladeia, does my answer help? Do you need any further assistance with this one?

Ladeia, 2018-01-16:
Sorry for delay, I migrate the project for LEDE OS and now I can compile the project :-)

closed 2018-01-16 by Ladeia

vanbwodonk, 2018-03-01:
Hi how do you compile it on LEDE? i try but i can make it works, help me please

this is my  feeds for linkit use LEDE, i can compile it without mraa & upm.
https://github.com/vanbwodonk/lks7688-LEDE-feeds

shmrymbd, 2019-02-27:
hi @Ladeia could you help how to compile lede with latest mraa and upm? @vanbwodonk