eclipse-mraa/mraa #851: Cant find libmraa-java in ubuntu packet installer

GMIT-mmurray, 2018-01-03:
Any ideas where I could libmraajava.so for ubuntu 
I have mraa.jar and the examples compile OK but I get the following error
when the programs runs ...

"Dynamic Linking Problems in the SWIG Java documentation for help.
java.lang.UnsatisfiedLinkError: no mraajava in java.library".

root@student-UP-CHT01:/home/student# sudo apt-get install libmraa1 libmraa-dev libmraa-java python-mraa python3-mraa node-mraa mraa-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libmraa-java
E: Unable to locate package node-mraa
root@student-UP-CHT01:/home/student#

arfoll, 2018-01-03:
did you do an ```apt-get update``` first? Looks like the repo isn't finding the packages at all which is odd. Can you paste your apt config and your OS version ```cat /etc/os-release```? In any case the libmraajava.so once installed will be in /usr/lib/x86_64-linux-gnu/. Now that I think about it wonder if anyone tests the 32bit pkgs anymore... Here's to hoping no one is stuck on 32bit these days :)

GMIT-mmurray, 2018-01-03:
I am using Intel UP board.
 
Yes I did the apt-get update

<!DOCTYPE busconfig PUBLIC
 "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <policy user="root">
    <allow own="org.debian.apt"/>
  </policy>
  <policy context="default">
    <allow send_interface="org.debian.apt"/>
    <allow send_interface="org.debian.apt.transaction"/>
    <allow send_destination="org.debian.apt"/>
  </policy>
</busconfig>

NAME="Ubuntu"
VERSION="14.04.5 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.5 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"

arfoll, 2018-01-04:
So the issue is that we don't support 14.04 with the PPA, you'll have to
build from soruce to get everything. There is a package that @tingleby did
but it's very limited in scope.

Upgrade to 16.04 would be the 'easy' advice.

Brendan

GMIT-mmurray, 2018-01-04:
Thank you Brendan, 16.04 it is and working.