eclipse-mraa/mraa #854: Any plans to support TinkerBoard S

mgiaco, 2018-01-17:
Hi,

Are there any plans to support the new Tinker Board S?

https://www.asus.com/Single-Board-Computer/Tinker-Board-S/#tinker-board-kv-info

many thanks
mathias

arfoll, 2018-01-17:
Hi Mathias,

Looks like a nice board, I wasn't aware of it - I'm not sure anyone has looked into adding support but we would happily help anyone do the platform support and include the patch into master without any reservations.

On a sidenote, this benchmark with 'competitor SBC' made me laugh :)
[image]

mgiaco, 2018-01-18:
okay the bad thing is it doesn´t have a PRU.

closed 2018-01-18 by mgiaco