eclipse-mraa/mraa #856: [UP2] SPI frequency is limited to 1 MHz

nukular0, 2018-01-19:
Hi,

This issue is basically the same as #255, except on the UP Squared:
The driver reports a incorrect value of 1 MHz for SPI_IOC_RD_MAX_SPEED_HZ and therefore the SPI frequency can't be set above this value.

The same workaround mentioned in #255 of commenting out the lines 
`// if (ioctl(dev->devfd, SPI_IOC_RD_MAX_SPEED_HZ, &speed) != -1) {`
`// if (speed < hz) {`
`// dev->clock = speed;`
`// syslog(LOG_WARNING, "spi: Selected speed reduced to max allowed speed");`
`// }`

in [spi.c](https://github.com/intel-iot-devkit/mraa/blob/master/src/spi/spi.c#L258) works, but a real fix schould probably be implemented as well (like https://github.com/intel-iot-devkit/mraa/commit/644064f883a47ab73010569d521031006076b8fe ?)

The guys from Emutex have been informed and [are looking into it](https://forum.up-community.org/discussion/comment/7788/#Comment_7788), but who knows when this will be fixed in the driver.

Also: I am not sure if this is the right place to mention this, but the 10 MHz max speed for SPI reported in the [docs](https://iotdk.intel.com/docs/master/mraa/up2.html) for the UP2 is not correct. Without the sanity check code above, I was able to get 50 MHz.

arfoll, 2018-01-19:
Thanks for notifying us - I've pushed 2 commits in spi_max_speed branch to try fix this. I think we should just move this check to simply be informational, it's quite clear that most drivers have this set completely wrong or are way too pessimistic, so let's leave the burden on the users reading their datasheets and not exceeding what manufacturers claims will work rather than banning them from doing it. I've also removed the spi_replace function from the rpi as with this change it's no longer needed.

If it passes travis I'll merge it asap.

alext-mkrs, 2018-03-04:
@arfoll, I see it still lives in the branch (I've added one comment there BTW) - do you still want to pursue this? If not, I can incorporate my comment and merge it.

arfoll, 2018-03-05:
Agreed - merged & added your suggestion

closed 2018-03-05 by arfoll