eclipse-mraa/mraa #864: BUILDSWIGPYTHON or BUILDSWIGNODE fails

Barleyman, 2018-02-06:
Compiling with UP2 with ubuntu 16.4 using freshly pulled MRAA library (Feb 6th)

[output.txt](https://github.com/intel-iot-devkit/mraa/files/1698713/output.txt)

I've got swig 3.08 installed:
`olli@LAB-UP2:~/Downloads/mraa-master/build$ apt search swig | grep swig | grep installed
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
swig3.0/xenial,now 3.0.8-0ubuntu3 amd64 [installed]
`

Compile works fine if both of those options are OFF in cmake-gui.

arfoll, 2018-02-06:
That looks weird - can you share your full cmake conf line and what happens on `cmake` run as well as the full output of `make -j1`?

Your swig version should be absolutely fine for all bindings unless you have node v6/v7 but even then this gVersion failure is weird, I've never seen anything like it.

Barleyman, 2018-02-06:
I was using cmake-gui actually. Well, after seeing some errors in cmake output, I determined I was missing two packages. 
sudo apt install nodejs-legacy python3-dev
took care of that. The previous is a known problem with using /usr/bin/nodejs instead of /usr/bin/node

Unfortunately no go..

`cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++ -DBUILDDOC=ON -DBUILD_SHARED_LIBS=OFF .. &> cmake.txt
`
[cmake.txt](https://github.com/intel-iot-devkit/mraa/files/1698998/cmake.txt)

[makelog.txt](https://github.com/intel-iot-devkit/mraa/files/1699000/makelog.txt)

arfoll, 2018-02-06:
Ah ok the issue is coming from build shared libs being disabled - turn that on and the bindings will work - any reason for wanting fully static bindings?

Barleyman, 2018-02-06:
I can build it static with the options in title off.. I have a problem with
first SPI transaction  taking excessive time, I was hoping a static library
would help. This happens *after* the library has been initialized by setup
routines. The program goes into loop of "process about 15ms, update,
process about 15ms, update, ..". Slowdown happens after every process
routine. Data is 96 bytes stored in memory updated to two different
devices, each getting 48 bytes.

Excessive in this case being 20-40ms while the immediately following
subsequent call to the same library takes about 2ms for several
transactions. It may be a flat out bug which I'm trying to debug at the
moment.

arfoll, 2018-02-06:
Ok - fair enough. So if you really want static binaries, you'd have to build a static python for example. You can build the python module statically but that's considered non portable see: https://cmake.org/pipermail/cmake/2007-May/014347.html. You can however force it to build by doing:

```
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_C_FLAGS=-fPIC ..
```

```
make _python3-mraa
```

```
brendan@schultz ~/git/mraa/build/src/python/python3 (git)-[master] % ldd _mraa.so 
        linux-vdso.so.1 (0x00007ffdb7de5000)
        libpython3.6m.so.1.0 => /usr/lib/libpython3.6m.so.1.0 (0x00007f114ad17000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f114aaf9000)
        librt.so.1 => /usr/lib/librt.so.1 (0x00007f114a8f1000)
        libjson-c.so.3 => /usr/lib/libjson-c.so.3 (0x00007f114a6e2000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f114a35b000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007f114a00f000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f1149df8000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007f1149a41000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f114983d000)
        libutil.so.1 => /usr/lib/libutil.so.1 (0x00007f114963a000)
        /usr/lib64/ld-linux-x86-64.so.2 (0x00007f114b4e3000)
```

Barleyman, 2018-02-06:
I have no need for python so that's not a problem! Maybe an idea to amend
the build docs about building static.

Barleyman, 2018-02-06:
Well, still no go. When I try to use the compiled libmraa.a I get a bunch of JSON error messages from the linker:
[foobar.txt](https://github.com/intel-iot-devkit/mraa/files/1699278/foobar.txt)

Barleyman, 2018-02-06:
Including pthread.h and json_object.h to the program source does nothing. In fact gpio.h already *has* pthread.h which the compiler then complains about.

arfoll, 2018-02-06:
You need to link your binary against json-c and pthread since you're using a static libmraa... Add ```-ljson-c -lpthread```.

Barleyman, 2018-02-06:
Thank you, I wouldn't have thought of that. Now it compiles no problem and .. there's absolutely no difference on that first SPI transaction lag. I have to dig deeper. 

To clarify, the state machine implements a simple protocol with the target microcontroller acting as a slave with a separate \SPI_READY signal indicating it's, well, ready. 

After a bit of timing tests, I confirmed it's indeed the first byte going out after a short delay that's causing excessive delay. Any ideas about that? Perhaps some other forum?
This state machine snippet takes 20-30ms to execute, if I run it without the debug flags it doesn't do it every other time, with the debug flags and timestamping in place it's every other time like a clockwork. I know because I measure the total transaction time separately.. 

`        if(arguments.debug){
        	clock_gettime(CLOCK_MONOTONIC, &delay1);
        }
        /*Poll if idle*/
        if(SPI_state.SPI_idle){
            *txbyte=ENQ;
            *rxbyte=POLL_SPI(SPI_packet.SPI_txbyte);
            SPI_state.SPI_idle=0;
            *delay=40;
            if(arguments.debug){
            	char chars;
            	clock_gettime(CLOCK_MONOTONIC, &delay2);
            	elapsed=(Time_To_Micros(&delay2)-Time_To_Micros(&delay1));
                chars=snprintf(loopstring, stringsize, "p%u,",elapsed);
                if (chars>stringsize) break;
                stringsize=stringsize-chars;
                loopstring=loopstring+chars;
            }
            continue;
        }`

Pushing out the byte, "side" is determined when parsing library arguments with an associated call to change parameters earlier:

`UCHAR Poll_SPI_Mraa(UCHAR foo){
    static UCHAR byte;
    if(arguments.side){
    	byte=mraa_spi_write(SPI_state.SPI1,foo);
    }
    else {
    	byte=mraa_spi_write(SPI_state.SPI0,foo);
    }
    return byte;
}`

Barleyman, 2018-02-06:
I also try to prevent swapping out and boost timings like so:
```
void LD_Set_Priority(){
    struct sched_param sp;
    struct rlimit rl;
    memset(&sp, 0, sizeof(sp));
    memset(&rl, 0, sizeof(rl));
    sp.sched_priority = sched_get_priority_max(SCHED_FIFO);
    sched_setscheduler(0, SCHED_FIFO, &sp);
    mlockall(MCL_CURRENT | MCL_FUTURE);
    getrlimit(RLIMIT_NICE, &rl);
    rl.rlim_cur=30;
    rl.rlim_max=30;
    setrlimit(RLIMIT_NICE, &rl);
}
```

closed 2018-02-07 by arfoll

Barleyman, 2018-02-07:
You also need to include the json-c and pthread libraries to make it work. For what it's worth, since there's no benefit to compiling mraa as a static library performance-vice, I've gone back to "normal" way of doing it as a shared library. 

By the way, the -fPIC flag not default for mraa? From what I gather the main difference is that PIC is somewhat slower to execute but not on the level where it would matter in this instance while "non-PIC" takes longer time to load but is faster after that.

arfoll, 2018-02-07:
Well that's normal - building a static libmraa doesn't include json-c & pthread statically. I wouldn't expect it to increase performance, -fPIC is not the default for static binaries since it's not needed if you do static builds - only reason we need it with bindings is because static + dynamic mixture. It's enabled in dynamic builds.

commit 568c0770 (2018-02-07): building.md: Add note for static builds and bindings

Closes #864