eclipse-mraa/mraa #865: ASpiDevice_transfer function?

Barleyman, 2018-02-06:
As I've had a problem with the first byte out of the SPI bus taking excessive time, I decided to look closer into where the magic happens. I traced the SPI transfer function to peripheralman.c, which calls for example ASpiDevice_transfer(dev->bspi, &data, &recv, 1)

So far so good but this function does not appear to exist anywhere.. Google search only finds mraa source plus one android project git. Is this a kernel driver function?

arfoll, 2018-02-07:
This function calls the google PIO/peripheral manager interfaces. If you're getting a delay here then likely it's because you're calling binder and your messages are taking time. Check: https://github.com/androidthings/native-libandroidthings/blob/master/x86/include/pio/spi_device.h#L69 and https://developer.android.com/things/sdk/pio/spi.html.

Barleyman, 2018-02-07:
WRT #866 stats, it does not happen every time for some reason and disabling PM gives some benefit.

Should it be beneficial to call that peripheral manager .transfer method directly bypassing the MRAA library? 

Is this PIO peripheral interface a standard kernel "thing" for 4.x or something you have to go out of your way to compile in? at least it seems to exist in Ubuntu 16.04. 

What about the kernel SPI support? https://www.kernel.org/doc/html/v4.14/driver-api/spi.html that google interface is a wrapper for it?

arfoll, 2018-02-07:
PIO is google android things specific. It will always introduce a delay as every call goes through an IPC. Mraa of course will add a small delay but compared to the IPC cost it's going to be near irrelevant.

If you're on ubuntu then you're not using PIO so I'm kinda confused... Maybe it would be beneficial if you explained your setup a little more (OS, kernel, stack), we may be able to advise you some more.

Barleyman, 2018-02-07:
Well:
`Linux LAB-UP2 4.10.0-42-generic #5000~upboard10-Ubuntu SMP Tue Dec 19 12:58:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

My colleague for some reason decided to set up this box with ubuntu instead of ubilinux. Perhaps the idea was to try development with "generic" linux to highlight problems customers might have using our library. But the setup is messed enough that I believe I'll take the nuclear option at this point and reflash ubilinux 4.

arfoll, 2018-02-07:
So in that case you're not using the peripheral manager code paths at
all... In fact that code isn't even being built... I would recommend going
back to your tracing and start again with your analysis.

Barleyman, 2018-02-07:
Is ubilinux on UP2 using this code? 

What is exactly required to use the peripheral manager code? I said before this looks like an Android thing..?

arfoll, 2018-02-07:
no it's not - you're digging in the wrong place. To use peripheralmanager you need to be on android things OS not ubuntu/ubilinux...

Barleyman, 2018-02-07:
All right.. So what is mraa library actually calling? As I said I worked
out  the spi transfer function redirects to this ASpidevice function but
apparently I made a mistake somewhere along the way. What is the transfer
function actually using?

arfoll, 2018-02-07:
So on UP2, the platform configuration src/x86/up2.c doesn't override any normal mraa functionalities so for spi look in src/spi/spi.c and whatever function you called in C will be there. If you called mraa_spi_transfer_buf then look at https://github.com/intel-iot-devkit/mraa/blob/master/src/spi/spi.c#L375. We're essentially doing an iotctl straight onto the kernel spidev file handle so you won't go any faster yourself (well maybe you could remove one func call and a bit of error handling) without basically making kernel level code.

alext-mkrs, 2018-02-07:
Let me close this one as we have established this is not a PeripheralManager-based setup and nothing in UP board-related code uses that. The issue of SPI delays will be dicsussed within #866

closed 2018-02-07 by alext-mkrs