eclipse-mraa/mraa #866: Delay with first byte out

Barleyman, 2018-02-06:
I have two SPI devices on two CS lines sharing the same bus. My program updates them  back to back in a loop with data processing happening in-between. The 2nd one does not appear to exhibit this behavior while the 1st one takes 20-40ms to send the first byte out.

It does not matter if I do mraa_spi_init_raw() & mraa_spi_stop() entering/exiting my SPI library or even if I try to be sneaky and issue init after the 2nd device is done before processing next set of data. 
Normally I'd just leave the SPI bus "open" until program exit. 

the init and stop functions do NOT cause appreciatable delays by themselves.

Propanu, 2018-02-06:
Have you tried turning off power management for the SPI devices?

> SPI PM can sometimes do weird things you can disable it with: `echo on > /sys/devices/pci0000\:00/0000\:00\:07.1/power/control`

On the UP2 the spi devices are 19.0 and 19.2.

Barleyman, 2018-02-07:
This does actually give some benefit but does not fix the problem entirely. 

Power control auto:
```Frame Time,Resize Time,LD Time,SPI TimeL, SPI TimeR,Display Time, Total Time
55,0,28,32,1,426,542
7,0,25,19,1,137,189
7,0,21,18,1,84,131
7,0,23,18,1,46,95
7,0,26,1,1,62,97
7,0,23,1,11,47,89
7,0,22,34,1,45,109
8,0,18,1,1,58,86
7,0,23,33,1,50,114
7,0,19,33,1,46,106
```
Power control on
```
Frame Time,Resize Time,LD Time,SPI TimeL, SPI TimeR,Display Time, Total Time
53,0,28,2,1,441,525
7,0,25,1,1,146,180
7,0,21,14,1,81,124
7,0,20,1,1,48,77
7,0,26,2,1,59,95
8,0,21,1,1,47,78
7,0,22,2,1,55,87
7,0,19,3,1,47,77
7,0,23,2,1,67,100
```

So basically lag occurrence is greatly reduced and the lag is halved to 14ms vs 18-34ms with power management enabled.

14ms is still unfortunately not quite good enough. nb display time is huge since I'm running this on remote desktop!

alext-mkrs, 2018-02-07:
@Barleyman, could please show the relevant piece of code where you communicate with those devices (the whole loop you're talking about)?

Barleyman, 2018-02-07:
@alext-mkrs The "loop" is in fact a state machine which is probably too extensive to include here, not to mention company ip.. But I I did in fact put the relevant pieces here: #864 

I'd say it's a given the "first" call to transfer a buffer on spi buffer is the culprit, given how I'm calling time before and after that exact bit and getting the (much) longer than expected delay. 

I nuked the ubuntu on the up squared board and I'm now setting up ubilinux environment on it. This will take a bit before I'm back "online". Probably have things working again tomorrow, I got clang/llvm 5.0 and Eclipse Oxygen set up properly now. Code is safely on company git.

alext-mkrs, 2018-02-10:
Ok, please post back when you have the OS replacement sorted out. In general, as you can see in mraa's `spi.c`, there's nothing in the library code, which would distinguish writes in such a [or any other] way. In `mraa_spi_write()` you use, we just allocate one SPI "transfer" structure and do an `ioctl()` call with that. So at the first glance that would seem to be kernel-level, but let's first see if the OS replacement fixes this right away and if not, we can dig deeper.

Barleyman, 2018-02-16:
FYI, I had a fresh setup on UP2 working last friday ready to actually test the software sans some bugfixing. Installing ubilinux was easy but the application uses OpenCV with hardware decoding and OpenGL so it was not that straightforward to setup from scratch. This time I documented the steps as I inherited the code. 

Then I was out of office for two days and some wags dismantled the UP^2 setup meanwhile.. So this is taking longer than it should.

alext-mkrs, 2018-02-17:
Thanks for the update and take your time - whenever you have information for us to troubleshoot this further and there's someone available to do that - we'll surely help you.

Barleyman, 2018-02-26:
Okay, finally got everything together again. I did some restructuring on the application front end meanwhile which shouldn't really have anything to do with this. But it was a horrible unstructured mess and I inherited the code so I rewrote it to remotely resemble a structured code.. 

Okay, so now I have a known setup on ubilinux - It seems the problem is most definitely power saving on this setup. Using @Propanu suggestion makes the problem go away. Although the PCI device for SPI #0 on UP squared is 0000:00:19.0, not 07.1. 

So, with power control on "auto", I get repeateable double digit millisecond delay on first SPI byte out after each processing cycle. So apparently the kernel decides to put the bus to sleep since it's not been accessed for dozen milliseconds or so. However, I still get outliers.

`Loop times:t0:l27,p37643,t37:l37737,t37:l37850,p321,t38:l38223,t38:l27,p336,t38:l413,t39:l908,p302,t39:l1288,t39:l1347,p269,t39:l1667,t40:l1726,p206,t40:l1971,t41:l34,p188,t41:l301,
Parse time 0ms, Execution time 41ms, loops: 14, errors: 0
Write OK
`
"p" in this case being "poll" in microseconds. 

And the same with control "on":
`Loop times:t0:l72,p9829,t10:l9968,t10:l10070,p146,t10:l10259,t10:l24,p165,t10:l225,t10:l409,t11:l443,p54,t11:l531,t11:l563,p55,t11:l651,t11:l684,t11:l716,p53,t11:l802,t12:l25,t12:l75,p81,t12:l189,
Parse time 0ms, Execution time 12ms, loops: 17, errors: 0
Write OK
`

With power control forced on, worst case delay is 10-20ms versus 20-40ms with power control on auto. It also happens (much) less often. "Normal" transaction is 2-3ms total so we're still looking at 5-10x outliers.

alext-mkrs, 2018-03-04:
Ok, this makes sense now. Power management has caused such effects for me personally in the past too (with SPI and UART devices). As for the remaining outliers - as we have a general-purpose OS here and not the RTOS with guaranteed syscall service times, I think that's the best you could get, without resorting to things like PREEMPT_RT patch, for example.

Taking those recent findings and my previous analysis, I don't think this is mraa-related and can be fixed or alleviated there - so let me close this one. If you need any further help, feel free to comment (or submit a new one, if the topic is unrelated) and we'll go from there.

closed 2018-03-04 by alext-mkrs

Barleyman, 2018-03-05:
I think a common workaround would be a keep-alive function to ping the slave every millisecond or so. An asynchronous update function running on it's separate thread should do it. I do have some higher priority items to concentrate on first, thought. Theoretically a daemon with DBUS connectivity would be ideal but getting around to that is "someday in the future". 

Does MRAA/UP squared play nice with the RT patch using "hard" pre-emptiveness? At least with RPi there are issues, bluetooth or wifi do not like it if my memory serves.