eclipse-mraa/mraa #867: Up Squared Board - unkown platform

mpetitjean, 2018-02-07:
Hi,

I am unable to use GPIOs with mraa on an Up Squared Board. I suspect the board is not properly recognized because running this C code

~~~~
#include "mraa.h"
int main(int argc, char** argv)
{
    const char* board_name = mraa_get_platform_name();
    fprintf(stdout, "hello mraa\n Version: %s\n Running on %s\n", mraa_get_version(), board_name);

    mraa_deinit();
    return MRAA_SUCCESS;
}
~~~~

yields 
~~~~
hello mraa
 Version: v1.8.0-40-g568c077
 Running on Unknown platform
~~~~

Any idea ? The output of `uname -a` on the board is `Linux upsquared 4.4.0-ubi4-amd64 #1 SMP Debian 4.4.35-1 (2016-11-29) x86_64 GNU/Linux`.

arfoll, 2018-02-07:
What's the output of:
```cat /sys/devices/virtual/dmi/id/board_name```

mpetitjean, 2018-02-07:
The output is `4293W1E`

arfoll, 2018-02-07:
Ok so apply this quick patch and it should work. However I have no idea why your board is responding differently to our boards over here - @nicola-lunghi @jarteaga ideas? Have you done a bios update? If you can past the output of:
```
cat /sys/devices/virtual/dmi/id/bios_version
cat /sys/devices/virtual/dmi/id/bios_date
```

```
diff --git a/src/x86/x86.c b/src/x86/x86.c
index 8836ff9..81a54c1 100644
--- a/src/x86/x86.c
+++ b/src/x86/x86.c
@@ -93,7 +93,8 @@ mraa_x86_platform()
             } else if (strncasecmp(line, "UP-CHT01", strlen("UP-CHT01") + 1) == 0) {
                 platform_type = MRAA_UP;
                 plat = mraa_up_board();
-            } else if (strncasecmp(line, "UP-APL01", strlen("UP-APL01") + 1) == 0) {
+            } else if (strncasecmp(line, "UP-APL01", strlen("UP-APL01") + 1) == 0 ||
+                       strncasecmp(line, "4293W1E", strlen("4293W1E") +1) == 0) {
                 platform_type = MRAA_UP2;
                 plat = mraa_up2_board();
             } else if (strncasecmp(line, "RVP", strlen("RVP") + 1) == 0) {
```

mpetitjean, 2018-02-07:
I'm sorry, I messed up and pasted the output of my laptop and not the board, silly me. The actual output is `UP-APL01`. It corresponds to the MRAA_UP2 then.

mpetitjean, 2018-02-07:
`bios_version` is `UPA1AM21`,
`bios_date` is `09/01/2017`

arfoll, 2018-02-07:
Makes more sense but fix is a bit more tricky then :) - Can you check syslog/journald so we can see what's going on? Mraa should output a bunch of info on there

mpetitjean, 2018-02-07:
Here are the logs corresponding to the hello C code above:

~~~~
Feb  7 11:21:14 upsquared libmraa[8388]: libmraa version v1.8.0-40-g568c077 initialised by user 'root' with EUID 0
Feb  7 11:21:14 upsquared libmraa[8388]: Adding i2c bus found on i2c-1 on adapter i2c_designware.1
Feb  7 11:21:14 upsquared libmraa[8388]: up2: kernel pinctrl driver unavailable
Feb  7 11:21:14 upsquared libmraa[8388]: up2: Platform failed to initialise
Feb  7 11:21:14 upsquared libmraa[8388]: libmraa initialised for platform 'Unknown platform' of type 98
~~~~

arfoll, 2018-02-07:
so for some reason the pinctrl driver isn't loaded as ```/sys/bus/platform/drivers/upboard-pinctrl``` doesn't seem accesible - have you tried loading it?

mpetitjean, 2018-02-07:
There might be the issue: the driver seems to be loaded as `/sys/bus/platform/drivers/up-pinctrl`.

arfoll, 2018-02-07:
Try this patch:
```
diff --git a/src/x86/up2.c b/src/x86/up2.c
index 4517362..5460ec5 100644
--- a/src/x86/up2.c
+++ b/src/x86/up2.c
@@ -251,7 +251,7 @@ mraa_up2_board()
     // Configure ADCs
     b->aio_count = 0;
 
-    const char* pinctrl_path = "/sys/bus/platform/drivers/upboard-pinctrl";
+    const char* pinctrl_path = "/sys/bus/platform/drivers/up-pinctrl";
     int have_pinctrl = access(pinctrl_path, F_OK) != -1;
     syslog(LOG_NOTICE, "up2: kernel pinctrl driver %savailable", have_pinctrl ? "" : "un");
```

We'll have to work out if this is a long term name and deal with it when the up2 guys respond.

mpetitjean, 2018-02-07:
Works like a charm now, both platform recognition and GPIO control. Many thanks to you !

arfoll, 2018-02-07:
Ok thanks alot - I'll wait for Aeeon to respond and see what they say reguarding the new name, we could just remove the check altogether and 'hope'...

nicola-lunghi, 2018-02-07:
Hi
the user is using the wrong kernel

Linux upsquared 4.4.0-ubi4-amd64 #1 SMP Debian 4.4.35-1 (2016-11-29) x86_64 GNU/Linux.

For the up squared the user should use either ubilinux 4 or ubuntu 16.04.01 with the 4.10 kernel from the ppa

So no changes are necessaries in mraa
Best Regards
Nicola Lunghi

ps by the way contact @Dan-Emutex about this I'm not on the project anymore :-)

jarteaga, 2018-02-08:
Hi all,

To expand a bit on @nicola-lunghi's answer, at the moment we ship **two** out-of-tree drivers in our kernel releases:
* `up-pinctrl` for the "legacy" setup (2016, supports UP1 with the initially-shipped BIOS and FPGA firmware)
* `upboard-pinctrl` for all other platforms (2017, supports UP2, UP Core and UP1 with updated BIOS+FPGA)

The kernel above is an older release that did not contain the newer `upboard-pinctrl` driver for UP2. Patch above will force detection of course, but I don't think GPIO control will fully work on the 40-pin header (just on the SoC end). Without UP2 FPGA support, you won't be able to change GPIO direction or pin mode from the default settings.

@arfoll: we're looking into upstreaming `upboard-pinctrl`, which will most likely require further changes after kernel maintainer feedback. MRAA UP1 support was sent in 2016, targeting the legacy `up-pinctrl` (which we won't attempt to upstream).

So we're hoping to finalise the MRAA platform detection code when we know what the upstreamed driver will look like. I can then send patches so MRAA correctly detects the platform in all cases. Am I correct to assume you would rather always stay compatible with the existing releases?

arfoll, 2018-02-08:
Thanks for replying - yes we'd like to stay compatible with everything, if this is only naming let's just check for either driver to be present.

jarteaga, 2018-02-08:
Well, it's not really just naming - `up-pinctrl` simply does not implement UP2 support. The pins can't be properly controlled in this setup (I'm guessing @mpetitjean did not need to change pin settings).

So the platform not being detected is warranted here. A kernel update fixes that.

arfoll, 2018-02-08:
Ah ok - makes sense. @mpetitjean all good?

closed 2018-02-08 by arfoll