eclipse-mraa/mraa #868: mraa on pico-imx6 / pico-imx7

TGunthorpe, 2018-02-10:
Has anyone install mraa on arm7 pico-imx6 or pico-imx7 ?

I've followed this procedure:
git clone https://github.com/intel-iot-devkit/mraa.git && cd mraa && \
mkdir build && cd build && \
cmake -DBUILDSWIGNODE=OFF -DBUILDSWIGPYTHON=ON -DCMAKE_INSTALL_PREFIX:H=/usr .. && \
make install

This successfully installs mraa version :  v1.8.0-41-g2d21062

My C/C++ application compiles with no errors, however at runtime when making hardware calls to 
UART, PWM or GPIO the initializations fail.

I've tested on kernel 4.1.15 and 4.9 with the same results.

any ideas?

thanks in advance

alext-mkrs, 2018-02-10:
As you can see in our [readme](https://github.com/intel-iot-devkit/mraa/blob/master/README.md), mraa doesn't formally support these SBCs/SOMs. However depending on the pinmuxing complexity, you can try to use the [JSON way](https://github.com/intel-iot-devkit/mraa/blob/master/docs/jsonplatform.md) of initializing the platform for these.

And if you decide to hack mraa to add proper support, patches/PRs would be more than welcome, we can help with that ;)

TGunthorpe, 2018-02-12:
Hi Alex,

 

I have a need to create mraa (1.8.0 or later) for pico-imx6 and pico-imx7.

 

Not sure where to start or what is required or how it’s done, but I’m happy to do the work.

 

If you could just give me a start, I’ll continue.

 

Best regards

Tom

alext-mkrs, 2018-02-12:
So I'd first suggest you to look into the JSON way of initializing platform - it may turn out there's nothing to be done, depending on the platform complexity. That link I posted earlier should get you started and feel free to ask if something is unclear, here or in the Slack/IRC/mail list.

If it's complex enough to warrant the traditional way of adding the support - I'd suggest you to start by looking at past platform addition PRs, see links below.

https://github.com/intel-iot-devkit/mraa/pull/832
https://github.com/intel-iot-devkit/mraa/pull/829
https://github.com/intel-iot-devkit/mraa/pull/818
https://github.com/intel-iot-devkit/mraa/pull/666

alext-mkrs, 2018-06-25:
Okay, so as this was a question and it was already answered - I'm closing this one. PRs with i.MX-based SBCs discussed above are still more than welcome! :)

closed 2018-06-25 by alext-mkrs