eclipse-mraa/mraa #869: cmake error on latest RASPBIAN STRETCH LITE

mgiaco, 2018-02-25:
Hi

Can someone help me here. I try to build mraa for a raspberry pi 1 but I get this error. Hm should I update cmake? 

```
pi@raspberrypi ~/python/libs/mraa/build $ cmake ..
-- INFO - libmraa Version v1.9.0
-- INFO - cmake Version 2.8.9
-- INFO - Target arch is armv6
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
INFO - Adding ARM platforms
CMake Warning at src/CMakeLists.txt:203 (message):
  Enabled JSONPLAT but JSON-C library not found: Ignoring JSONPLAT

-- PYTHON2 attempting to build!
CMake Error at src/python/python2/CMakeLists.txt:13 (target_include_directories):
  Unknown CMake command "target_include_directories".

-- Configuring incomplete, errors occurred!
```

arfoll, 2018-02-25:
yeah if you don't have target_include_directories cmake is too old or disable python bindings, though does show a bug, we should change minimum version to 2.9.

mgiaco, 2018-02-26:
okay many thanks.

closed 2018-02-26 by mgiaco