eclipse-mraa/mraa #870: Compile on LEDE cant find head-ref

vanbwodonk, 2018-03-02:
Hi i try to run make libmraa for LEDE on linkit MT7688.
I try to compile release v1.9.0.ta.gz
But i got this error?

```
CMake Error at CMakeFiles/git-data/grabRef.cmake:36 (file):
  file failed to open for reading (No such file or directory):

    /home/drw/LINKIT/source-17.01.4/build_dir/target-mipsel_24kc_musl-1.1.16/libmraa-1.9.0/CMakeFiles/git-data/head-ref
```

It seems cmake detect build from git source, how i can bypass git check when compiling mraa?

vanbwodonk, 2018-03-02:
If i add empty file head-ref inside libmraa-1.9.0/CMakeFiles/git-data/ I can compile it successfully

arfoll, 2018-03-02:
delete things around https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L90 and just keep https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L94.

If that's seriously an issue then we can probably add a -DGITVERSIONING flag to disable that cmake module.

vanbwodonk, 2018-03-02:
> delete things around https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L90 and just keep https://github.com/intel-iot-devkit/mraa/blob/master/CMakeLists.txt#L94.

@arfoll Thanks for response, try your suggestion and it works perfect. I will create patch for that on monday. This is my Makefile https://github.com/vanbwodonk/lks7688-LEDE-feeds/blob/master/libmraa/Makefile

> If that's seriously an issue then we can probably add a -DGITVERSIONING flag to disable that cmake module.

This is will perfect. I search probably any option like this in docs. But i can find it.

arfoll, 2018-03-05:
Could you try the following to see if it fixes it? I'm curious and can't seem to reproduce when this is exactly happening:

```
diff --git a/cmake/modules/GetGitRevisionDescription.cmake.in b/cmake/modules/GetGitRevisionDescription.cmake.in
index 888ce13..82a4237 100644
--- a/cmake/modules/GetGitRevisionDescription.cmake.in
+++ b/cmake/modules/GetGitRevisionDescription.cmake.in
@@ -32,7 +32,7 @@ else()
        configure_file("@GIT_DIR@/HEAD" "@GIT_DATA@/head-ref" COPYONLY)
 endif()
 
-if(NOT HEAD_HASH)
+if(NOT HEAD_HASH AND EXISTS "@GIT_DATA@/head-ref")
        file(READ "@GIT_DATA@/head-ref" HEAD_HASH LIMIT 1024)
        string(STRIP "${HEAD_HASH}" HEAD_HASH)
 endif()
```

tcpipchip, 2018-03-31:
Hi, have you the patch or new Makefile that works ?