eclipse-mraa/mraa #871: Up Squared - Nodejs on Ubuntu: How to compile?

lenhuy, 2018-03-02:
1. Test.js
var m = require('mraa');
console.log("MRAA version: " + m.getVersion());
var digitalPin = new m.Gpio(12);

2. node Test.js

MRAA version: v1.9.0
Test.js:3
var digitalPin = new m.Gpio(12);
                         ^
Error: Illegal arguments for construction of _exports_Gpio

3. sudo node Test.js

module.js:328
    throw err;
    ^

Error: Cannot find module 'mraa'

Propanu, 2018-03-02:
Use `sudo -E node Test.js`, or modify your `sudoers` file to keep `NODE_PATH`.

lenhuy, 2018-03-02:
1. sudo -E node Test.js
Got the same error same "node Test.js"

MRAA version: v1.9.0
Test.js:3
var digitalPin = new m.Gpio(12);
                         ^
Error: Illegal arguments for construction of _exports_Gpio

2. How do I modify sudoers file to keep NODE_PATH?

alext-mkrs, 2018-03-04:
@lenhuy, you can find many tutorials/guides on configuring `sudo` using your favorite search engine, as this is a standard Linux tool. It's strange that `sudo -E` didn't work. Please post the contents of your system syslog/journal around the time you run this, mraa should be logging more details there.

Propanu, 2018-03-05:
Are you using the Ubuntu [image](https://downloads.up-community.org/download/up-squared-iot-grove-development-kit-ubuntu-16-04-server-image/) provided by Emutex for the UP2, or just plain vanilla Ubuntu? The latter won't give you access to the GPIO pins as it doesn't have the required kernel driver.

Propanu, 2018-03-12:
You can find instructions on how to patch the kernel here: https://wiki.up-community.org/Ubuntu

lenhuy, 2018-03-14:
I'd would like to use GrovePi board for the Up Squared, which packets that I need to install?

lenhuy, 2018-03-16:
I got it working now. Need the line below to interface with Grove Sensors.
mraa.addSubplatform(mraa.GROVEPI, "0 ")

closed 2018-03-16 by lenhuy