eclipse-mraa/mraa #875: Incorrect UART device on UP2 board

malikabhi05, 2018-03-10:
The UART device path in mraa for the up2 board is hardcoded to "ttyS1" (in https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/up2.c). The holds true for ubilinux, however i have an up2 board running the following:
BIOS: UPA1AM21
OS: 4.10.0-42-generic #5000~upboard10-Ubuntu SMP Tue Dec 19 12:58:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

and the UART device on the up2 that i have is available on ttyS5. 
As per https://wiki.up-community.org/Pinout_UP2, the M10 uart device node number can be checked here:
/sys/bus/pci/devices/0000\:00\:18.0/dw-apb-uart.8/tty/

and the HAT header uart device node number here:
/sys/bus/pci/devices/0000\:00\:18.1/dw-apb-uart.9/tty/

Maybe the uart device path being set in up2.c should be decided based on which node number is present in the paths mentioned above. Maybe an approach similar to the one employed to resolve the i2c bus numbers (mraa_find_i2c_bus_pci in mraa.c) could be used for the uart devices here.

arfoll, 2018-03-13:
yup, very good idea - patches welcome :)

alext-mkrs, 2018-03-21:
As there's now PR #878 opened for this - let me close this one.

closed 2018-03-21 by alext-mkrs