eclipse-mraa/mraa #877: sensor read with mraa

cristina21r, 2018-03-14:
I want to make an opc-ua server on galileo gen 2, with a yocto built version of linux.
I've made, with some help from git, a complete version of that kind of server for raspberry pi 3, but i have some problems with porting it to galileo. I'm using for the server dht11 and ds18b20 sensors, but i can't manage to make it work on galileo. 

Here is my code:

var opcua = require("node-opcua");
var w1bus = require("node-w1bus");
var bus = w1bus.create();
var rpiDhtSensor = require("rpi-dht-sensor");

function read_sensor(sensor,callback) {
                bus.getValueFrom(sensor)
                .then(function(res){  callback(null,res.result); })
                .catch(callback);
        };

function start_server() {

        var server = new opcua.OPCUAServer({
                port: 1234
        });

        server.buildInfo.productName ="Rapsberry OPCUA Server"

        function post_initialize() {

                var addressSpace = server.engine.addressSpace;
                var devices = addressSpace.addFolder("ObjectsFolder",{ browseName: "Devices"});
                var temperatureSensor = "28-041460c2a3ff";
                install_sensor(temperatureSensor);
        };

        var tempValue = -20.0;
        var tempValue1 = -20.0;
        var humValue = -20.0;

        function install_sensor(sensor) {
                var addressSpace = server.engine.addressSpace;
                var sensorName = "ds18b20";
                var sensorName1 = "dht11";
                var devices = addressSpace.rootFolder.objects.devices;
                var sensor  = addressSpace.addObject({
                        organizedBy: devices,
                        browseName:  sensorName
                });

                var dht = new rpiDhtSensor.DHT11(27);

                setInterval(function() {
                        var temperatureSensor = "28-041460c2a3ff";
                        read_sensor(temperatureSensor,function(err,result) {
                                tempValue = result.value;
                        });
                        var readout = dht.read();
                        tempValue1 = readout.temperature.toFixed(2);
                        humValue = readout.humidity.toFixed(2);
                },3000);

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature1",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "temperature2",
                        dataType:   "Double",
                        value: {
                                get: function () {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: tempValue1 });
                                }
                        }
                });

                addressSpace.addVariable({
                        propertyOf: sensor,
                        browseName: "humidity",
                        dataType: "Double",
                        value: {
                                get: function() {
                                        return new opcua.Variant({dataType: opcua.DataType.Double, value: humValue });
                                }
                        }
                });
        };

        server.initialize(post_initialize);

        server.start(function() {
                console.log("\n\nTemperature - OPCUAServer");
                var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
                console.log("Server endpoint url is ", endpointUrl );
                console.log("\n");
        });
}

start_server();

I want to read info from one of these sensors, or others, if easier,with mraa, in order to use those info for my opcua server..
How can I read the temperature from sensors?? Pleasee, help..

arfoll, 2018-03-14:
What galileo image are you using? It's pretty unsupported at this stage so I can't recommend it. Generally we do not support the use of the DHT11 sensor as it's essentially bit banged and very hard to use reliably.

cristina21r, 2018-03-14:
Yocto 3.8.7..
I don't need to do that just with dht11 sensor.. ds18b20 or any other example would be just fine..

arfoll, 2018-03-14:
I think you mean linux kernel 3.8.7 - FYI that's really really old.

The DS18B20 is a 1-wire sensor which we do support via uart emulation. But again that's not the kind of sensor we'd recommend. If you want cheap use an analog sensor, if you want something good there are a number of good i2c & spi sensors available readily.

For the DS18B20 (or any other sensor where you want a premade lib) use UPM, see the documentation there: http://upm.mraa.io/sensorDetail.html?name=DS18B20

alext-mkrs, 2018-03-17:
@cristina21r, I see you've opened a UPM issue for further discussion about ds18b20 communication, so I'll close this one.

closed 2018-03-17 by alext-mkrs