eclipse-mraa/mraa #879: mraa mock platform interacts with underlying hardware

malikabhi05, 2018-03-15:
The mraa mock platform interacts with the underlying hardware while calling the function `mraa_get_number_of_gpio_chips()` in `src/gpio/gpio.c:mraa_gpio_chardev_init`, returning the number of gpio chips available on the underlying hardware for the mock platform.

And then eventually further down  `mraa_get_chip_info_by_number()` fails returning NULL.

alext-mkrs, 2018-03-21:
That's an interesting one - sounds like when adding those chardev-related functions necessary _replace overrides were not added for the mock platform. I'll look into this on the weekend if I have time.

Propanu, 2018-03-21:
No need to, my chardev patch should fix this.

Propanu, 2018-03-24:
@malikabhi05 does this work as expected now?

malikabhi05, 2018-04-01:
yes, it seems to work as expected now. closing it.

closed 2018-04-01 by malikabhi05