eclipse-mraa/mraa #880: MRAA - SWIG 4.0 - .NET Core 2.1 C# - arm64v8

IvanFarkas, 2018-03-21:
Would it be possible to include **SWIG 4.0** to generate **.NET Core 2.1 C#** to use **MRAA** on **arm64v8**, please?
Has anything done in this area?
If not, how can I help, please?

arfoll, 2018-03-21:
Well I know very little about .net but youd need to add a swig interface
file and that's no problem..

As for arm64v8 I don't have many arm targets but feel free to add some to
arm/ I don't believe you need to change anything for arm64v8 specifically?

alext-mkrs, 2018-03-21:
@IvanFarkas, and what's your end goal, which specific .NET implementation (.NET Core, Mono)?

As for arm architecture specifically - as @arfoll said, it's a bit irrelevant and orthogonal to the language bindings. Architecture comes into play when we are talking about specific target boards you want to run mraa on. Which one you have in mind here?

Propanu, 2018-03-21:
Due to the similarities between C# and Java, I think the best approach to this would be to look at how we currently generate our Java bindings and go from there. SWIG will play the biggest role here. Is .NET Core a subset of .NET for Windows IoT Core? Sorry, I haven't kept up with .NET for some time, but the idea of developing a C# app with MSVS and running it on a Linux target with Mono/.NET Core is very enticing.

IvanFarkas, 2018-03-21:
Thank you very much @arfoll @alext-mkrs @Propanu for the quick feedback.
The target board is **arm64v8 DragonBoard 410c, DragonBoard 820c**.
I like the **Java to C# conversion**. That's what I would've done as well.

alext-mkrs, 2018-03-22:
Let's continue in #778, so that we have only one thread going.

@propanu, .NET Core is Microsoft's alternative and open-source implementation of the .NET Standard - [here's](https://msdn.microsoft.com/en-us/magazine/mt842506.aspx) a nice general overview.

closed 2018-03-22 by alext-mkrs

IvanFarkas, 2018-03-22:
Sounds great

arfoll, 2018-03-23:
little title change since at first I thought this was about SWIG 4.0 :D

IvanFarkas, 2018-03-23:
Make sense. Thx.

deinok, 2018-09-21:
https://www.nuget.org/packages/SystemExt.Devices.Mraa/

IvanFarkas, 2018-09-21:
Awesome! I need it on Linux arm64 on DragonBoard 820c using .NET Core 2.2 C#.
Is it available for Linux arm64 (Debian)?
Perhaps GitHub source?

deinok, 2018-09-21:
https://github.com/SystemExt/SystemExt.Devices.Mraa
It should work on any device that can run netstandard1.1 and have a `libmraa` installed.
In the near future you are not going to need libmraa installed in the SO, It will use a precompiled version attached to the nuget package

deinok, 2018-09-23:
@IvanFarkas Now we have a way to include MRAA without with just adding a NuGet Package. It doesn't require having it installed.