eclipse-mraa/mraa #883: Building mraa for Linkit Smart 7688 error： node.h not found and v8.h and uv.h not found

harveypang, 2018-03-31:
building error message：
CMake Warning at CMakeLists.txt:93 (message):
   - Install git to compile a production libmraa!

-- INFO - libmraa Version v1.8.0
-- INFO - cmake Version 3.7.1
-- INFO - Target arch is mips
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
INFO - Adding MIPS platforms
PYTHON2 attempting to build!
ERROR - node.h not found
ERROR - v8.h not found
ERROR - uv.h not found
CMake Error at /home/pang/Work/LEDE_17.01/staging_dir/host/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find Nodejs (missing: NODEJS_INCLUDE_DIRS)
Call Stack (most recent call first):
  /home/pang/Work/LEDE_17.01/staging_dir/host/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindNodejs.cmake:58 (find_package_handle_standard_args)
  src/javascript/CMakeLists.txt:1 (find_package)

but I can find the files: 
/usr/include/nodejs/src/node.h
/usr/include/nodejs/deps/uv/include/uv.h
/usr/include/nodejs/deps/v8/include/v8.h
What can I do to avoid this mistake? Please help me.
ubuntu 16.04 building LEDE17.01 for linkit smart 7688

malikabhi05, 2018-04-01:
could you please share the cmake command that you're running. Also what OS are you using?

alext-mkrs, 2018-04-04:
@malikabhi05, the path from the error log suggests it's LEDE (aka OpenWRT): https://en.wikipedia.org/wiki/LEDE

@harveypang indeed the command you run would be useful, as the log looks strange - there should be a message produced by

```cmake
message(STATUS "INFO - NODE_ROOT_DIR is ${NODE_ROOT_DIR}")
```

before the `ERROR - node.h not found` one.

Could you in addition to the command itself post a full cmake output?

alext-mkrs, 2018-06-10:
@harveypang, is this still actual? Please provide the requested information if yes.

alext-mkrs, 2018-06-25:
Closing for no response.

closed 2018-06-25 by alext-mkrs