eclipse-mraa/mraa #884: UpSquared up2 not enough current passing through gpio

Nyceane, 2018-03-31:
Hey guys, the gpio works, but when i connect it to a relay via D7 on grove shield, it would unlock the deadbolt on Arduino but wouldn't do it via up2 board, any ideas on how to fix it?

alext-mkrs, 2018-04-04:
How do you think this could be related to mraa library, I wonder? :)

I'd suggest you to start with cross-checking the parameters or your relay vs parameters of the GPIOs provided by the Grove shield. If that's the grove relay you're talking about, I'd expect that to work and then please post the code you have + describe what exactly do you observe. That's still highly unlikely to be related to the library, but just for the fun of it, why not...

alext-mkrs, 2018-06-10:
@Nyceane, is this still relevant?

alext-mkrs, 2018-06-17:
Okay, but that again sounds like the HW part of your setup. Have you tried what I suggested in the previous comment? That simply sounds like your relay isn't compatible with your board's GPIO electrical params, that's it.

Anyway, as this sounds to be solved - I'm closing this one. Let us know if any further help is needed.

closed 2018-06-17 by alext-mkrs