eclipse-mraa/mraa #885: Up2 board, ValueError: Error initialising UART

Nyceane, 2018-03-31:
Can not run the sample UART files...

Initializing UART...Traceback (most recent call last):
  File "uart_sender.py", line 34, in <module>
    u = mraa.Uart("0")
  File "/usr/lib/python3.5/dist-packages/mraa.py", line 1061, in __init__
    this = _mraa.new_Uart(*args)
ValueError: Error initialising UART

malikabhi05, 2018-04-01:
so a couple of quick questions on this. What OS are you using ubilinux/ubuntu? and also did you get from the ppa or did you build it on your system? Also whats your mraa version?

alext-mkrs, 2018-04-04:
+ please post the syslog/journal output around the time you run this - mraa may be logging additional useful information there.

malikabhi05, 2018-04-06:
@Nyceane #878 This commit actually resolves an issue similar to what you're seeing. Because this fix is fairly recent, you might not have it if you installed mraa from ppa. However, if you build mraa locally, you should be able to get UART to work on the UP2 board.

alext-mkrs, 2018-06-10:
@Nyceane, is this still actual?

Nyceane, 2018-06-17:
yes, using Ubuntu

alext-mkrs, 2018-06-17:
What about other questions? That information is necessary to understand the situation and help you.

alext-mkrs, 2018-06-25:
So @Nyceane, are you up to providing the information requested? Without it we can't help you any further.

alext-mkrs, 2018-07-02:
Closing for no reply.

closed 2018-07-02 by alext-mkrs