eclipse-mraa/mraa #890: Help wanted

MagdyA, 2018-04-08:
Can someone tell me what are the prerequisites and steps needed to install latest mraa and upm on intel galileo with yocto image on it?

I could install it once 2 months ago with an old read me file on the repo which had clearer steps for installing.

things got changed and I can't remember how I did it so I tried as docs says in here
[image]
but I get error in command **make** that I must use c99 flag and here is the full error message I get on the terminal

https://pastebin.com/guDYxNua

I'm a total newbie so pardon me for any stupid step I made.

arfoll, 2018-04-12:
Your compiler is *old* so default C standard is also old. Add -DCMAKE_C_FLAGS="-std=gnu99" to your cmake line (make sure to wipe the build/ dir before hand).

I made a branch c99x with a probable fix and will probably merge something like that in next few days. Other option is just to compile latest stable afaik that still works.

MagdyA, 2018-04-13:
Sounds like it. Thanks for your support

alext-mkrs, 2018-06-10:
Closing as clarified.

closed 2018-06-10 by alext-mkrs