eclipse-mraa/mraa #891: GrovePi Shield I2C/GPIO Conflict

srware, 2018-04-10:
Using the GrovePi shield with the Up2 board, all libraries up to date.

When I have the Temperature & Humidity I2C sensor connected (SI7005 provided with the Grove Developer Kit) I get an error when I attempt to use any of the GPIO through MRAA.

To test I connect the SI7005 to any of the I2C ports on the GrovePi shield. If I now initialise and attempt to write or read from a GPIO port I get a -1 return and I2C error. If I unplug the SI7005 sensor it immediately starts working.

Ideas?

srware, 2018-04-12:
So it looks like there are some conflicting I2C register addresses between the SI7005 and GrovePi, in particular the GPIO read and write addresses are also used for the SI7005. I don't know if the SI7005 has configurable addresses which can be assigned when the module is initialised but this is probably a feature request for UPM, not MRAA.

closed 2018-04-16 by srware

Propanu, 2018-04-17:
This is unfortunate, but it's caused by devices not being I2C compliant, thus there's not a lot we can do from software. SI7005 is all fixed, GrovePi could in theory use a different firmware that avoids this.