eclipse-mraa/mraa #893: Revert PR #886

Mani-Sadhasivam, 2018-04-11:
@malikabhi05 You should revert this PR. The commits are messed up and it needs to be resolved. Ivan pushed the changes to one of my previous commit #887 . I am planning to do resolve commits and open a new PR with Ivan's changes.

malikabhi05, 2018-04-11:
@Mani-Sadhasivam Could you provide some clarity on this, I just want to understand this better. Since we cannot rewrite history, are you asking that we revert 284d882b9db8637dc67773a86e9cf4a8564230a3 and 7ea32cf1ecce38d3ecf245d5245f244390ab0a86 or either one of them? This would remove support for Dragon board 820c added by @IvanFarkas . Is there anything in one of Ivan's commits which is wrong and shouldn't be there? or is it just that one of the commits in his PR was based on one of your older commits and the commit message got carried over to the new commit? 

@pylbert @Propanu thoughts?

IvanFarkas, 2018-04-11:
@malikabhi05 

1. Please think about all those people who you **deprive** by delaying this long-awaited change.
2. Regrettably, you are **not allowed** using any of my code without my **permission**.
3. **Be positive**, please.

Propanu, 2018-04-11:
I think what happened was that both @Mani-Sadhasivam and @IvanFarkas worked on the same source file at the same time and the changes weren't rebased properly, it happens. The issue I see with it is [here](https://github.com/intel-iot-devkit/mraa/commit/284d882b9db8637dc67773a86e9cf4a8564230a3#diff-da50b7193d3f8af5271bd9eb7aa32f04L76), the chardev map for the hikey board is removed. And 7ea32cf is a partial duplicate of dee7b80 that @IvanFarkas picked up in his PR. I assume it was unintentional, and all that's needed is a bit of clean-up.

Mani-Sadhasivam, 2018-04-12:
@malikabhi05 Exactly it is same of what @Propanu said. I'm happy with @IvanFarkas change of adding DB820c. But somewhat the commits needs a bit of cleanup. If you are good to do then please go ahead and clean the commits. But it is completely unintentional from @IvanFarkas and all I want is a clean commit history.

Mani-Sadhasivam, 2018-04-12:
@malikabhi05 There should be two commits:

1. Adding basic board support for DB820c
2. Adding Chardev support for DB820c

Both were already added by @IvanFarkas but not organised properly in commits.

arfoll, 2018-04-12:
@IvanFarkas 
1. Aren't you being a little dramatic here?
2. I'm not sure what you mean by this? Please read - https://github.com/intel-iot-devkit/mraa/blob/master/CONTRIBUTING.md
3. Sorry but I'm not seeing what you're seeing - who isn't being positive? We screwed up, no one is debating that, let's just figure out how to fix it asap...

@malikabhi05 this doesn't rewrite history it's not rebasing which as you say we can't do now it's on master. Personally I'm not a huge fan of reverts because it just adds garbage to the history forever, but it does mean that someone needs to clear up the mess in either case. @malikabhi05 since it's your mess I propose you start, please treat this as urgent as it is not good to leave master in this state.

IvanFarkas, 2018-04-12:
**Leaving to Budapest Hungary today.**
**Be back in early May.**
I'll have limited Internet access.
I am visiting my friends and family who I have not seen for almost 2 years.

I agree with your rules.
If you decide to revert, please **allow me to add my code later, so I can get the credit**.
Thank you very much for your understanding in this matter,

IvanFarkas, 2018-04-12:
No.
It works perfectly.
I tested it for GPIO A-L on my DB820c.

malikabhi05, 2018-04-12:
Thanks @IvanFarkas for testing this and confirming that it works. We can't re-write history and reverting it isn't such a good option. Apart from the incorrect commit message, there seems to be nothing broken or out of place(specific to the files changed in the 2 commits). @Mani-Sadhasivam can we close this?

Mani-Sadhasivam, 2018-04-13:
@malikabhi05 @pylbert There is no issue with the implementation related to `src/arm/96boards.c` but the issue is with the commits adding them. 

Commit https://github.com/intel-iot-devkit/mraa/commit/7ea32cf1ecce38d3ecf245d5245f244390ab0a86 should only add DB820c board support(there should be no chardev support for DB820c/Hikey). For commit https://github.com/intel-iot-devkit/mraa/commit/dee7b80f063f869bc393544a57e177dc070d6d03 it should only add DB820c chardev mapping not Hikey. It is already mentioned by @Propanu. So if you guys don't care about the commit history then I have no issues, else someone has to do this little cleanup.

But again, there is nothing wrong with the implementation (couple of pin mappings are wrong which I'll fix it)

Mani-Sadhasivam, 2018-04-16:
Hi @malikabhi05 @Propanu Any update on this issue?

Propanu, 2018-04-16:
Hi @Mani-Sadhasivam, considering only 1 commit got messed up and since functionality wise everything is in order, I would leave this as is. I know it's not ideal and it's totally on us, we'll try to be more careful when rebasing/merging in the future. We have new PR's open that are waiting to be merged so we really need to move on.

Mani-Sadhasivam, 2018-04-17:
@malikabhi05 @Propanu Thanks for taking your time to look into this issue. As we all agreed to move on, closing this issue.

closed 2018-04-17 by Mani-Sadhasivam