eclipse-mraa/mraa #900: Segmentatio fault. Intel EDISON - custom board

elgarbe, 2018-04-29:
I'm trying to get mraa working on my board, I've installed jubilinux and the follow instructions from:
https://learn.sparkfun.com/tutorials/installing-libmraa-on-ubilinux-for-edison

When I try blink led I get segmentation fault.

The same thing happen if I try to import mraa into python:

```
edison@jubilinux:~$ sudo ./blink 
Segmentation fault
edison@jubilinux:~$ nano blink.c
edison@jubilinux:~$ python
Python 2.7.13 (default, Jan 19 2017, 14:48:08) 
[GCC 6.3.0 20170118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa as m
Segmentation fault
```
/var/log/messages:

```
Apr 29 22:46:02 localhost libmraa[16912]: libmraa version v1.9.0-32-gb59fc8f initialised by user 'root' with EUID 0
Apr 29 22:46:02 localhost kernel: [ 2843.950705] blink[16912]: segfault at 3b0 ip b76f6aaf sp bfc46a00 error 4 in libmraa.so.1.9.0[b76ee000+39000]
Apr 29 23:04:19 localhost libmraa[16994]: libmraa version v1.9.0-32-gb59fc8f initialised by user 'edison' with EUID 1002
Apr 29 23:04:19 localhost kernel: [ 3940.575937] python[16994]: segfault at 3b0 ip b6987aaf sp bf815270 error 4 in libmraa.so.1.9.0[b697f000+39000]

```

elgarbe, 2018-05-01:
It works on ubilinux, but not in jubilinux. why?

alext-mkrs, 2018-06-10:
So th esecond run from the log was with non-root user, that won't work without messing up with udev/device file permissions, so use root.

The first run contains surprisingly low amount of information. Looks like it's crashing somewhere *very* early, as supported by the python run. Could you try running this under `strace` to see more details of what's happening? If you're able to run under GDB and get a stack trace, that would be even more useful.

jannsta1, 2018-06-10:
Coincidentally I've just followed the same procedure as @elgarbe and suffered the same problem. Attached is the console output from an attempt to import mraa in a python script when running under strace.
[strace_mraa_import_seg_fault_sudo.txt](https://github.com/intel-iot-devkit/mraa/files/2088046/strace_mraa_import_seg_fault_sudo.txt)

alext-mkrs, 2018-06-10:
Thanks. So two things I see are:
1. Some SELinux-related stuff is apparently being loaded. Please disable SELinux temporarily (or make sure it's disabled);
1. There's an error message being written to syslog [I presume], apparently the sudo is not providing us the root we want it to:
```
stat64("/usr/bin/sudo", {st_mode=S_IFREG|S_ISUID|0755, st_size=163988, ...}) = 0
write(2, "sudo", 4sudo)                     = 4
write(2, ": ", 2: )                       = 2
write(2, "effective uid is not 0, is /usr/"..., 133effective uid is not 0, is /usr/bin/sudo on a file system with the 'nosuid' option set or an NFS file system without root privileges?) = 133
write(2, "\n", 1
```

jannsta1, 2018-06-11:
Thanks @alext-mkrs for looking at the trace. I realised after looking at the bits that you mentioned that the problem in my case was that I have various python installations (because of my ROS installation) and I hadn't provided root access for the python version I was importing mraa with.

Is there a way to configure mraa to run without root access? That would really save a lot of headache with configuring and using the python bindings. Or are there any other recommended python libraries for accessing Edison GPIO that don't require root access? 

An obvious feature request is to handle this seg fault with a warning in the python binding.

alext-mkrs, 2018-06-13:
So as far as root is concerned - its possible, but not easy and essentially boils down to changing permissions on device files mraa uses. But then there are problems with diing this generically and dynamically - there was a discussion on this in one of the opened issues, I don't have a link OTMH, but there wasn't a full solution. Generally speaking that's beyond what mraa as a library can do and  is a matter of OS config.

As for the handling in python binding - it's generated automatically, so we can't modify it. That's why the root requirement is documented, that was the best we could do at the time.

alext-mkrs, 2018-06-13:
that's the one I was talking about, you can try applying the approach discussed there: https://github.com/intel-iot-devkit/mraa/issues/802

jannsta1, 2018-06-13:
Thanks for the response and the link. Looks like mraa outside of sudo is still work in progress but will subscribe to that issue for updates. I kind of hijacked this issue (#900) but it seems like the likely cause is now known.

alext-mkrs, 2018-06-13:
Right :) Glad to help, anyway ;)

@elgarbe, so if you still need help, feel free to chime in. If not, I'll close this one in a few days.

Propanu, 2018-06-13:
@jannsta1 one alternative would be to take a look at how Emutex does it on the UP2 board, I'm sure the udev rules can be made to work with the Edison too: https://wiki.up-community.org/Ubuntu#Enable_the_HAT_functionality_from_userspace
Here are the actual sources for the rule files: https://launchpad.net/~ubilinux/+archive/ubuntu/up/+sourcefiles/upboard-extras/0.1-1/upboard-extras_0.1.orig.tar.gz

alext-mkrs, 2018-06-25:
Closing for no further activity.

closed 2018-06-25 by alext-mkrs