eclipse-mraa/mraa #903: MRAA number mapping for Intel Edison Arduino

lhy-loveworld, 2018-06-06:
Hi, I found [this](https://learn.sparkfun.com/tutorials/sparkfun-inventors-kit-for-edison-experiment-guide/appendix-e-mraa-pin-table) webpage which also gives a mapping from MRAA number to Arduino. And it's different from the one in the docs in this repository. For example, in this repo, A0's MRAA number is 14 however in that link, it is 31. I'm totally confused.

I have 2 questions about the mapping:
1. How to correspond the MRAA number to the one marked on Arduino?
2. If I want to use I2C on Arduino, what number should I pass to mraa_i2c_init()? And I found on Arduino, the SCL is connected to A5, and the SDA is connected to A4. Is it I2C-1 or I2C-6?

Thanks!

alext-mkrs, 2018-06-06:
So to be honest I don't get the idea of that table on SparkFun, but then again - I haven't analyzed it for too long. I'd suggest you to start with https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md, that's the best source for mraa pin numbering, if you use the Arduino board.

In essence, mraa's philosophy is to follow numbering that's printed on boards themselves and it's true for Edison Arduino board. So Arduino pin 3 would be pin 3 for mraa as well for all places that matter for you as a user.

As for i2c, the Arduino board exposes only i2c-6, so you should use 6 for init. However you can actually use anything - mraa will get an error internally when trying to initialize anything but 6 and will default to 6.

Let me know if you have any further questions.

lhy-loveworld, 2018-06-06:
@alext-mkrs Thank you for your response. So the MRAA number in this page https://github.com/intel-iot-devkit/mraa/blob/master/docs/edison.md is not the number printed on boards (which is the number  used by the initialization functions), right?

Propanu, 2018-06-06:
That table lists the SoC pins for the Edison module not the breakout boards. The Edison board definition file will actually detect if you have the mini or Arduino breakout and assign the pins to match the order on the corresponding board.

alext-mkrs, 2018-06-07:
Right, basically that big table is in the "breakout board" section, which means the so-called mini-breakout, which is ~a size of Edison and has all the Edison module's pins exposed. The "Arduino" board section doesn't contain any tables because the numbers are printed on the board itself and you can just use them straight away (there are just some clarifications regarding e.g. analog pins and I2C).

As it looks like the question was answered - I'm closing this one. Feel free to comment if you need further help.

closed 2018-06-07 by alext-mkrs