eclipse-mraa/mraa #905: can't install mraa

Dhiaaeddinanabtawi, 2018-06-21:
Hi, 

I am trying to install mraa using: npm install mraa, but I am getting this error message:
```
npm ERR! Linux 4.15.0-23-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "mraa"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.9.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.9.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

```
Well, I am sure I have the last version of both node.js and npm, so I think the problem lies somewhere else. I am new to this, so I am not able to figure it out my self. Any suggestions?

Thanks in advancd
Dhiaa

alext-mkrs, 2018-06-21:
> I am sure I have the last version of both node.js

...and that's the problem - see #598 and https://github.com/swig/swig/pull/968 - SWIG doesn't support Node versions > 6 at the moment, so you'd need to downgrade your Node to be able to use mraa with it, as our JS bindings are generated via SWIG.

Propanu, 2018-06-21:
If you want to use Node 8, you'll have to patch swig and build mraa from source. It's pretty straight forward and I've added instructions in the documentation a while back.

closed 2018-06-25 by alext-mkrs