eclipse-mraa/mraa #910: Error : Invalid GPIO pin specified  | Java | ARM | Debian

joepol, 2018-07-02:
Hi , I build the mraa project for an ARM architecture and Java binding .
The build was successful and so was compiling the Example.java and GpioRead6.java. 
when running "Example" I get : 

> hello mraa 
> Version: v1.9.0 
> Running on Unknown platform

**when running GpioRead6 I get an IllegalArgumentException : Invalid GPIO pin specified .**
tried #353  and #529 so
I tested the GPIO in terminal: 
```
echo 6 > /sys/class/gpio/export
echo in > /sys/class/gpio/gpio6/direction
cat /sys/class/gpio/gpio6/value
```
and it works - I manage to read it's value. 

but I still get the same error when using Java code. 
Any suggestions ?
Thanks.

alext-mkrs, 2018-07-02:
> Running on Unknown platform

That's the reason - mraa doesn't recognize the board it runs on, so it doesn't have GPIO maps etc. What is that you're running it on?

Propanu, 2018-07-02:
As @alext-mkrs mentioned an unknown platform will have no listed pins, however you can still initialize GPIOs in raw mode (by sysfs #) for testing. Check the Java API docs [here](https://iotdk.intel.com/docs/master/mraa/java/classmraa_1_1_gpio.html).

joepol, 2018-07-03:
@alext-mkrs  - it's some unknown type of the arm family, but I have the GPIO map. 
@Propanu  - Great! I used the raw mode and it works. 
Thank you.

closed 2018-07-03 by joepol