eclipse-mraa/mraa #911: Problem building MRAA python docs using latest python-sphinx v1.7.6 package

b49020, 2018-07-26:
Hi,

I have Debian buster running on my dragonboard 410c with latest python-sphinx v1.7.6 package installed. Following is sequence of instructions and error message observed:

```
$ git clone https://github.com/intel-iot-devkit/mraa.git -b v1.9.0
$ cd mraa/
$ mkdir build
$ cd build/
$ cmake .. -DBUILDDOC=ON -DBUILDSWIGPYTHON=ON -DBUILDSWIGNODE=OFF -DINSTALLGPIOTOOL=ON
```

Error message while configuration:
```
sphinx-build 1.7.6
CMake Error at src/python/python2/docs/CMakeLists.txt:41 (message):
  ERROR - Failed to find a compatible version of Sphinx.  Python API doc will
  not be generated

-- Configuring incomplete, errors occurred!
```
Any help regarding above issue is much appreciated.

Regards,
Sumit

b49020, 2018-07-27:
It seems that display format of version info changed from sphinx v1.3.5 to v1.7.6 as follows:
```
$ sphinx-build --version
In case of v1.3.5: "Sphinx (sphinx-build) 1.3.5"
In case of v1.7.6: "sphinx-build 1.7.6"
```
Due to this parser written in "cmake/modules/FindSphinx.cmake" file doesn't seems to work correctly.

alext-mkrs, 2018-07-29:
Thanks @b49020 for reporting this and finding the reason. Are you up to creating a PR to fix this, now that we have the root cause?

b49020, 2018-07-30:
@alext-mkrs Sure I would like to contribute. So I did tried to fix this configuration check for sphinx v1.7.6. But when I try to build MRAA further, I got following error:
```
/root/mraa/src/python/python2/mraa2.i:6: Error: Unable to find 'common_hpp_doc.i'
/root/mraa/src/python/python2/mraa2.i:7: Error: Unable to find 'gpio_class_doc.i'
/root/mraa/src/python/python2/mraa2.i:8: Error: Unable to find 'i2c_class_doc.i'
/root/mraa/src/python/python2/mraa2.i:9: Error: Unable to find 'pwm_class_doc.i'
/root/mraa/src/python/python2/mraa2.i:10: Error: Unable to find 'aio_class_doc.i'
/root/mraa/src/python/python2/mraa2.i:11: Error: Unable to find 'spi_class_doc.i'
/root/mraa/src/python/python2/mraa2.i:12: Error: Unable to find 'uart_class_doc.i'
src/python/python2/CMakeFiles/gen_python2-mraa_mraa2PYTHON_wrap.dir/build.make:61: recipe for target 'src/python/python2/mraa2PYTHON_wrap.stamp' failed
make[2]: *** [src/python/python2/mraa2PYTHON_wrap.stamp] Error 1
make[2]: *** Deleting file 'src/python/python2/mraa2PYTHON_wrap.stamp'
CMakeFiles/Makefile2:832: recipe for target 'src/python/python2/CMakeFiles/gen_python2-mraa_mraa2PYTHON_wrap.dir/all' failed
make[1]: *** [src/python/python2/CMakeFiles/gen_python2-mraa_mraa2PYTHON_wrap.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```
Do you have any clue for this build error?

alext-mkrs, 2018-08-01:
I haven't built docs much to be honest, but looking at the CMakeLists.txt and that mraa2.i file in the python2 dir, this looks rather like it's not finding the autogenerated files, which are included if the DOXYGEN define is found. @b49020 could you attach a full log, including the initial CMake run?

alext-mkrs, 2018-08-08:
Hmm, okay - so the log looks generally ok to me, nothing apparently wrong... No further ideas at this point, sorry. Generally the docs are being build-tested in our CI, so there probably is some peculiarity w.r.t. the tool versions you have, but I can't think of anything specific.

Are these files created in the build dir at all, could you check?

pylbert, 2018-08-08:
Version of CMake?  I believe the latest couple versions change some of the python target generation. Possible that the target dependencies are not creating the .i files prior to the swing call that uses them.

b49020, 2018-08-09:
@pylbert `-- INFO - cmake Version 3.11.2`

b49020, 2018-08-09:
@alext-mkrs No these *.i files aren't created in build directory.

arfoll, 2018-08-22:
Try make the doxygen docs first and make sure you wiped the build/ dir completely, BUILDDOCS is a bit fragile and there may be an order issue somewhere? Also install graphviz to get dot, I'm not sure it'll do much but it's never been tested without.

Try build this target for example:
$ make gpioclass_doc_i

You should then get a file in build/src/gpio_class_doc.i

b49020, 2018-08-22:
@arfoll Thanks for the suggestion. It seems to be order/dependency issue only. Using following commands, I am able to build successfully:
```
$ sudo apt install graphviz
$ cd mraa/
$ mkdir build
$ cd build/
$ cmake .. -DBUILDDOC=ON -DBUILDSWIGPYTHON=ON -DBUILDSWIGNODE=OFF -DINSTALLGPIOTOOL=ON
$ make common_hpp_doc_i
$ make gpioclass_doc_i
$ make i2cclass_doc_i
$ make pwmclass_doc_i
$ make aioclass_doc_i
$ make spiclass_doc_i
$ make uartclass_doc_i
$ make
```
Do you have any idea to resolve this order issue?