eclipse-mraa/mraa #915: ValueError: Error initialising UART

vjvicky007, 2018-08-10:
I have some error in IoT2000 model IoT2040 UART communication using python MRAA library,
please anyone help me to resolve this error.... 

root@iot2000:~# sudo python recv.py 
Traceback (most recent call last):
  File "recv.py", line 5, in <module>
    u = mraa.Uart(0)						   
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1521, in __init__
    this = _mraa.new_Uart(*args)
ValueError: Error initialising UART
root@iot2000:~#

alext-mkrs, 2018-08-12:
@vjvicky007 please post the contents of the syslog (`/var/log/syslog` or `journalctl -x` output, depending on what Siemens has there these days), there should be some additional information in there. Right now it looks like some UART-related error, which for Quark should not theoretically happen as it's been around for quite a while.

alext-mkrs, 2018-09-02:
Closing for no reply - feel free to respond if you still need assistance.

closed 2018-09-02 by alext-mkrs