eclipse-mraa/mraa #916: UP Squared Board - Problem using Grove Button on GrovePI - Configured as MRAA_GPIO_IN

BinduRao2018, 2018-08-10:
Hello,
I am using GrovePI on UP Squared Board. Instead of running the examples available with ArduinoCreate for GrovePI,  i am handcoding them using mraa/upm APIs.
I see the GPIO configured as IN doesn't seem to work properly on Digital Pins.

Here is the code snippet:
_...
mraa_add_subplatform(MRAA_GROVEPI, "0");
dev=mraa_gpio_init(BUTTON_PIN);
mraa_gpio_dir(dev, MRAA_GPIO_IN);
while (true)
{
   fprintf(stdout, mraa_gpio_read(dev));
}
..._
where BUTTON_PIN = 516 (connected to D4), which works.

Problem scenario:

1. Grove Button connected to D7, BUTTON_PIN set to 519  : Doesn't detect the key press/release
2. Grove Button connected to D4, BUTTON_PIN set to 519, key press/release is detected.

i.e independent of the 'pin_no' passed to the initialization, the key press/release is detected only on D4 (516).

alext-mkrs, 2018-08-12:
@BinduRao2018 could you please post the whole source code file? This looks a bit fishy - are you sure you aren't overriding `BUTTON_PIN` somewhere else, so it effectively points to D4 all the time?

BinduRao2018, 2018-08-13:
Yes, I am not overriding BUTTON_PIN anywhere. I have a simple example to read the GPIO for the GroveButton. Here is the complete source code for this:
------------------------------------------------------------------------------------------
_#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <stdlib.h>
#include <unistd.h>
#include "mraa/gpio.h"

//512 offset + digital pin 4
#define BUTTON_PIN 516_

int main(void)
{
    int mraa_return = 0;
    mraa_gpio_context dev;

    mraa_add_subplatform(MRAA_GROVEPI, "0");
    dev = mraa_gpio_init(BUTTON_PIN);
    if (dev == NULL) 
    {
      fprintf(stderr, "Failed to initialize BUTTON_PIN");
      mraa_deinit();
      return EXIT_FAILURE;
    }
    mraa_return = mraa_gpio_dir(dev, MRAA_GPIO_IN);
    if (mraa_return != MRAA_SUCCESS)
    {
       fprintf(stderr, "unable to set the gpio to input mode");
       goto err_exit;
    }
   
    while (1)
    {
       fprintf (stdout, "value = %d\n", mraa_gpio_read(dev));
       upm_delay(2);
    }

err_exit:
    mraa_result_print(mraa_return);
    mraa_deinit();
    return EXIT_FAILURE;
    
}_
------------------------------------------------------------------------------------------

BinduRao2018, 2018-08-13:
Update -

The problem no longer exists. Grove Button seems to work fine on any digitial pin. 

Observation:
When the problem was observed, there were some GPIO - 516 logs present in systemd, which is no longer logged now.
Can I assume it to be a power glitch??
Now in the logs, I can see _i2c5:write_byte_data: Access error: Remote I/O error_ from libmraa. Would this be an issue for the device connected on i2c?

BinduRao2018, 2018-08-17:
Thanks alex for the inputs. 
I am using the UP2 with the power adapter for UK as the kit came with the standard US power plug.  
I have now stopped ArduinoConnector.service as I don't use it and I don't see any errors on GPIO or I2C after that. 
I assume that this service was trying to initialize some of the pins by default, resulting in the error.