eclipse-mraa/mraa #917: mraa_gpio_lookup can't  check all physical pins names

turmary, 2018-08-14:
https://github.com/intel-iot-devkit/mraa/blob/d24457f9a15f5ad4470d55bc69fbece22ce97bfc/src/mraa.c#L920
When plat->gpio_count is less than plat->phy_pin_count, the function can't check pins between [ plat->gpio_count, plat->phy_pin_count );

I think it should be changed like this:
@@ -880,7 +880,8 @@ mraa_gpio_lookup(const char* pin_name)
         return -1;
     }

-    for (i = 0; i < plat->gpio_count; i++) {
+    for (i = 0; i < plat->phy_pin_count; i++) {

Mani-Sadhasivam, 2018-08-14:
@turmary `mraa_gpio_lookup` is supposed to check only GPIO pins not the physical pins exposed on the board. So, limiting the count to `plat->gpio_count` is fine here.

turmary, 2018-08-14:
@Mani-Sadhasivam 
OK, If it check GPIO only,  in the loop, the pin[i] type should be checked.
Make it has GPIO function or else don't incremment index variable i.

turmary, 2018-08-14:
Aug 14 12:25:26 raspberrypi libmraa[20886]: platform_detected = 9
Aug 14 12:25:26 raspberrypi libmraa[20886]: libmraa initialised for platform 'Raspberry Pi 3 Model B' of type 5
Aug 14 12:25:39 raspberrypi libmraa[20886]: [0] pin_name = 'GPIO12' plat_pins_name = 'INVALID'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [1] pin_name = 'GPIO12' plat_pins_name = '3V3'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [2] pin_name = 'GPIO12' plat_pins_name = '5V'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [3] pin_name = 'GPIO12' plat_pins_name = 'SDA0'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [4] pin_name = 'GPIO12' plat_pins_name = '5V'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [5] pin_name = 'GPIO12' plat_pins_name = 'SCL0'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [6] pin_name = 'GPIO12' plat_pins_name = 'GND'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [7] pin_name = 'GPIO12' plat_pins_name = 'GPIO4'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [8] pin_name = 'GPIO12' plat_pins_name = 'UART_TX'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [9] pin_name = 'GPIO12' plat_pins_name = 'GND'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [10] pin_name = 'GPIO12' plat_pins_name = 'UART_RX'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [11] pin_name = 'GPIO12' plat_pins_name = 'GPIO17'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [12] pin_name = 'GPIO12' plat_pins_name = 'GPIO18'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [13] pin_name = 'GPIO12' plat_pins_name = 'GPIO27'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [14] pin_name = 'GPIO12' plat_pins_name = 'GND'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [15] pin_name = 'GPIO12' plat_pins_name = 'GPIO22'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [16] pin_name = 'GPIO12' plat_pins_name = 'GPIO23'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [17] pin_name = 'GPIO12' plat_pins_name = '3V3'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [18] pin_name = 'GPIO12' plat_pins_name = 'GPIO24'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [19] pin_name = 'GPIO12' plat_pins_name = 'SPI_MOSI'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [20] pin_name = 'GPIO12' plat_pins_name = 'GND'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [21] pin_name = 'GPIO12' plat_pins_name = 'SPI_MISO'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [22] pin_name = 'GPIO12' plat_pins_name = 'GPIO25'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [23] pin_name = 'GPIO12' plat_pins_name = 'SPI_CLK'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [24] pin_name = 'GPIO12' plat_pins_name = 'SPI_CS0'
Aug 14 12:25:40 raspberrypi libmraa[20886]: [25] pin_name = 'GPIO12' plat_pins_name = 'GND'
Aug 14 12:46:09 raspberrypi libmraa[19018]: libmraa version v1.9.0-1-gce748f6 initialised by user 'pi' with EUID 1000

Mani-Sadhasivam, 2018-08-14:
@turmary Agree with you! There has to be a check for GPIO capability inside `mraa_gpio_lookup`. This will avoid unnecessary comparisons.

Will submit a PR for that! Thanks for bringing this up.

Mani-Sadhasivam, 2018-08-14:
@turmary Sorry, I was wrong. The count should be extended to `plat->phy_pin_count` since we have to check all the pins. Will change it. Thanks once again!

Mani-Sadhasivam, 2018-08-14:
Hi @turmary ,

I have submitted the fix #918. Can you please let me know if it addresses your issue or not?

turmary, 2018-08-14:
It's OK I think.thanks.

closed 2018-08-14 by turmary