eclipse-mraa/mraa #920: Invalid GPIO

psctec, 2018-08-26:
Traceback (most recent call last):
  File "main.py", line 9, in <module>
    green = mraa.Gpio(6)
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 752, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified

Linux Xubuntu 18.04 LTS

alext-mkrs, 2018-08-26:
Hi @psctec, this is not enough information to see what's going on, so could you please clarify:

1. What board this is?
1. Which mraa version is this
1. What is logged in syslog when you get this error?

psctec, 2018-08-26:
is an notebook lg-gram, i'm trying to put to work an fingerprint scanner FPC 1020.
mraa version 1.9
syslog:
Aug 26 20:01:09 gram libmraa[14464]: libmraa version v1.9.0 initialised by user 'xxx' with EUID 1000
Aug 26 20:01:09 gram libmraa[14464]: Platform not supported, not initialising
Aug 26 20:01:09 gram libmraa[14464]: imraa: Failed to open lock file
Aug 26 20:01:09 gram libmraa[14464]: libmraa initialised for platform 'Unknown platform' of type 98
Aug 26 20:01:09 gram libmraa[14464]: gpio: init: pin 6 beyond platform pin count (0)