eclipse-mraa/mraa #921: MRAA Pins on Pi 3 B+

Chuckduey, 2018-08-27:
I am using the latest Raspberry Pi 3 B+ that came out March 14 2018.  I downloaded MRAA 1.9 from Github and built it successfully.  The nodejs section was skipped due to another issue.  All the C and Python functions are working fine, except on the newest Pi 3 B+ it only shows 26 pins.  Where as the Slightly older Pi 3 (no plus) shows the 40 pins.  In looking into it further it appears that some update needs to be done in the raspberry_pi.c needs to be updated with the latest version of Pi. In looking at the PLATFORM area there is no PI3_B plus and in /proc/cpuinfo it comes up with:
Raspberry Pi 3 B+
Hardware: BCM2835
Revision	: a020d3
Serial	: 000000003c861ba1
MRAA Version Shows:
~$ mraa-gpio version
Version v1.9.0-60-g77d4676 on Raspberry Pi Model B Rev 1

Where as Pi 3 comes up with:
Raspberry Pi 3 B
Hardware: BCM2835
Revision	: a02082
Serial	: 00000000909161cf
and MRAA Version Shows:
~$mraa-gpio version
Version v1.9.0-60-g77d4676 on Raspberry Pi 3 Model B

In digging though the code it appears that if there is an unknown revision of hardware, the default is to call it a Pi Model B that only has 26 Pins. It looks like it is easy to fix by adding a020d3 with Raspberry Pi 3 B+  to the raspberry_pi.c code.  It would be nice to have it fixed up in Github for others to use.

Chuckduey, 2018-08-30:
OK, I did some code modifications can can get the Pi 3 B+ working right.  I added the PI_3_B_PLUS defines and added the case in both the /proc/cpuinfo and /proc/device-tree/compatible version check areas.  (Here is the /proc/cpuinfo check)
`               } else if (strstr(line, "a020d3")) {
                    b->platform_name = PLATFORM_NAME_RASPBERRY_PI3_B_PLUS;
                    platform_detected = PLATFORM_RASPBERRY_PI3_B_PLUS;
                    b->phy_pin_count = MRAA_RASPBERRY_PI3_B_PLUS_PINCOUNT;
                    peripheral_base = BCM2837_PERI_BASE;
                    block_size = BCM2837_BLOCK_SIZE;
                } else { `
And the /proc/device-tree/compatible additions:
`        } else if (mraa_file_contains(compatible_path, "raspberrypi,3-model-b-plus")) {
            b->platform_name = PLATFORM_NAME_RASPBERRY_PI3_B_PLUS;
            platform_detected = PLATFORM_RASPBERRY_PI3_B_PLUS;
            b->phy_pin_count = MRAA_RASPBERRY_PI3_B_PLUS_PINCOUNT;
        }`
 
The in the define area the PLUS platform was added and the raspberry_pi.h MRAA_RASPBERRY_PI3_B_PLUS_PINCOUNT was added.  

The code was compiled and installed.  Everything is working now:
 $ mraa-gpio version
Version v1.9.0-60-g77d4676 on Raspberry Pi 3 Model B+

 $ mraa-gpio list
01         3V3:
02          5V:
03        SDA0: GPIO I2C
04          5V:
05        SCL0: GPIO I2C
06         GND:
07       GPIO4: GPIO
08     UART_TX: GPIO UART
09         GND:
10     UART_RX: GPIO UART
11      GPIO17: GPIO
12      GPIO18: GPIO PWM
13      GPIO27: GPIO
14         GND:
15      GPIO22: GPIO
16      GPIO23: GPIO
17         3V3:
18      GPIO24: GPIO
19    SPI_MOSI: GPIO SPI
20         GND:
21    SPI_MISO: GPIO SPI
22      GPIO25: GPIO
23     SPI_CLK: GPIO SPI
24     SPI_CS0: GPIO SPI
25         GND:
26     SPI_CS1: GPIO SPI
27       ID_SD:
28       ID_SC:
29      GPIO05: GPIO
30         GND:
31      GPIO06: GPIO
32      GPIO12: GPIO
33      GPIO13: GPIO
34         GND:
35      GPIO19: GPIO
36      GPIO16: GPIO
37      GPIO26: GPIO
38      GPIO20: GPIO
39         GND:
40      GPIO21: GPIO

If there is interest in this code I can do a Pull Request and share this with everyone.

alext-mkrs, 2018-09-02:
@Chuckduey, yes, as long as we currently don't support that model and you seem to have added the necessary definitions, we'd be glad to have a PR for that.

Chuckduey, 2018-09-04:
OK, PR #924 Has the 2 files needed to add the Pi 3 B+ to the MRAA Libs so it is recognized correctly.

closed 2018-09-04 by Chuckduey