eclipse-mraa/mraa #925: UP2 board - A device connected to read GPIO (configured as IN), doesn't work when another device is connected on I2C

BinduRao2018, 2018-09-06:
I have connected Grove Button (GPIO) and Grove Temp & Humidity Sensor (I2C) on UP2 Grove board. I notice the following error in systemd, when I try to read GPIO. 

_*_Sep 05 06:20:49 ubuntu sudo[3355]: upsquared : TTY=pts/1 ; PWD=/home/upsquared/GrovePi/mraa_impln/grove-lightsensor ; USER=root ; COMMAND=./grove-lightsensor_*
*_Sep 05 06:20:49 ubuntu grove-lightsensor[3356]: calling mraa_init_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: libmraa version v1.9.0 initialised by user 'root' with EUID 0_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: testing log functionality_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: up2: kernel pinctrl driver available_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: libmraa initialised for platform 'UP2' of type 16_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: calling mraa_init_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: i2c_init: Selected bus 0_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: mraa: Added GrovePi subplatform_*
*_Sep 05 06:20:49 ubuntu libmraa[3356]: aio: Using sub platform_*
*_Sep 05 06:20:50 ubuntu libmraa[3356]: i2c5: write: Access error: Connection timed out_*
*_Sep 05 06:20:50 ubuntu libmraa[3356]: grovepi: failed to write command to i2c bus /dev/i2c-5_*
*_Sep 05 06:20:53 ubuntu libmraa[3356]: i2c5: write: Access error: Connection timed out_*
*_Sep 05 06:20:53 ubuntu libmraa[3356]: grovepi: failed to write command to i2c bus /dev/i2c-5_*
*_Sep 05 06:20:56 ubuntu libmraa[3356]: i2c5: write: Access error: Connection timed out_*
*_Sep 05 06:20:56 ubuntu libmraa[3356]: grovepi: failed to write command to i2c bus /dev/i2c-5_*
*_Sep 05 06:20:59 ubuntu libmraa[3356]: i2c5: write: Access error: Connection timed out_*
*_Sep 05 06:20:59 ubuntu libmraa[3356]: grovepi: failed to write command to i2c bus /dev/i2c-5_*_

Because of these errors,GPIO read return invalid value. When I disconnect the device from I2C,mraa_gpio_read() return correct values.
Is there any I2C interference with GPIO/AIO on UP2 board?

Propanu, 2018-09-06:
Hi @BinduRao2018, if you're using the GrovePi+ shield and the TH02/SI7005 temperature and humidity sensor then yes, the i2c bus will lose arbitration and eventually time out when other sensors are connected. This is likely caused by the shield's firmware, thus you might want to try updating that if a newer version is available. Other I2C temperature sensors (e.g. Bosch ones) should work fine though.

BinduRao2018, 2018-09-06:
Thanks @Propanu for quick response. Could you let me know on the way to check the GrovePI firmware version please?

Propanu, 2018-09-06:
Haven't seen any updates since 1.2.7 thus you're likely on the latest firmware already. You can follow the GrovePi+ project here: https://github.com/DexterInd/GrovePi. There are some examples and documentation on how to interact with the shield with i2c-tools which would be the easiest way to check the firmware version on it.

BinduRao2018, 2018-09-13:
I have checked with Dexter Industries as well, but they are not aware of any problem with the Grove sensor. I will check on Grove PI+ on Raspberry PI (using mraa) and get back to isolate the problem.

BinduRao2018, 2019-01-21:
@Propanu, Could you let me know the Bosch make for Temperature & Humidity instead of TH02/SI7005 to use on GrovePI please?
Is the sensor I2C based? Please provide the details