eclipse-mraa/mraa #929: Failing Building Mraa

Paraphase, 2018-09-12:
I'm trying to Install mraa on the Intel Edison running Miniconda with Python 3.7 
I Clone the Repository, create the Build folder and execute following command:
`cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_BUILD_TYPE=DEBUG -DJSON-C_INCLUDE_DIR=/home/root/miniconda3/envs/test/include/json-c/ ..`

When running the Make command, i get the following Error:

>/home/root/mraa/src/json/jsonplatform.c:31:25: fatal error: json-c/json.h: No such file or directory
> #include <json-c/json.h>
                         ^
> compilation terminated.
> src/CMakeFiles/mraa.dir/build.make:231: recipe for target 
> 'src/CMakeFiles/mraa.dir/json/jsonplatform.c.o' failed
> make[2]: *** [src/CMakeFiles/mraa.dir/json/jsonplatform.c.o] Error 1
> CMakeFiles/Makefile2:104: recipe for target 'src/CMakeFiles/mraa.dir/all' failed
> make[1]: *** [src/CMakeFiles/mraa.dir/all] Error 2
> Makefile:140: recipe for target 'all' failed
> make: *** [all] Error 2

The error occurs with or without the `-DJSON-C_INCLUDE_DIR=/home/root/miniconda3/envs/test/include/json-c/` option

Any Help please ?

Propanu, 2018-09-16:
Hi @Paraphase, for the Edison you don't need JSON platforms or the library so it's likely easiest to simply turn the option off completely with `-DJSONPLAT=OFF`.

Paraphase, 2018-09-18:
Hi Propanu
Thanks for your reply, i'll test it tomorrow

closed 2018-10-03 by Paraphase