eclipse-mraa/mraa #930: i2c0: write: Access error: Remote I/O error on Raspberry PI when mraa_gpio_write() is called

BinduRao2018, 2018-09-13:
Hi,
I am using GrovePI+ kit on Raspberry PI. I have written a small program to set a LED, which is connected to GPIO port (Port D4 on Grove PI+). This program works fine when run on UP2 IoT Development Kit, however give Access error on Raspberry PI.

Following updates were done to build and run an example:
- For Raspberry PI, libmraa is built by setting the  compilation options:
      -DBUILDARCH=arm -DBUILDSWIG=OFF.
- Updated the boot configuration to set I2C0 to on. i.e dtstatus=i2c0=on. (Linux pi3-2 4.14.62-v7+ available on PI)

I tried executing the application as root, but that doesn't help either.

Here is the logs that I got by running journalctl:

_Sep 13 10:44:59 pi3-2 sudo[793]:       pi : TTY=pts/0 ; PWD=/home/pi/mraa_tests/grove-led ; USER=root ; COMMAND=/bin/su
Sep 13 10:45:39 pi3-2 libmraa[824]: libmraa version v2.0.0-1-g3e7a468 initialised by user 'root' with EUID 0
Sep 13 10:45:39 pi3-2 libmraa[824]: gpio: platform doesn't support chardev, falling back to sysfs
Sep 13 10:45:39 pi3-2 libmraa[824]: libmraa initialised for platform 'Raspberry Pi Model B Rev 1' of type 5
Sep 13 10:45:39 pi3-2 libmraa[824]: i2c_init: Selected bus 0
Sep 13 10:45:39 pi3-2 libmraa[824]: mraa: Added GrovePi subplatform
Sep 13 10:45:39 pi3-2 libmraa[824]: gpio516: initialised on sub platform 'grovepi' physical pin: 4
Sep 13 10:45:39 pi3-2 libmraa[824]: i2c0: write: Access error: Remote I/O error
Sep 13 10:45:39 pi3-2 libmraa[824]: grovepi: failed to write command to i2c bus /dev/i2c-0
Sep 13 10:45:44 pi3-2 libmraa[824]: i2c0: write: Access error: Remote I/O error
Sep 13 10:45:44 pi3-2 libmraa[824]: grovepi: failed to write command to i2c bus /dev/i2c-0
Sep 13 10:45:46 pi3-2 libmraa[824]: i2c0: write: Access error: Remote I/O error
Sep 13 10:45:46 pi3-2 libmraa[824]: grovepi: failed to write command to i2c bus /dev/i2c-0_

BinduRao2018, 2018-09-17:
I see some changes (pull request - #938) for this version of Raspberry PI.  Is the problem mentioned above, related to the fix available? Please let me know.

Chuckduey, 2018-09-18:
Sorry about that, some one asked me for my BeagleBone code.  I will revert the beglebone.c back to the master and I will do a separate fork for the BeagleBone stuff and if I will do a separate PR.

Best regards

Chuck

BinduRao2018, 2018-09-18:
Thanks for the update @Chuckduey 
However, I am not clear about the solution. Apologies, if my question has created some confusion. With mraa(v 2.0.0),  I have a problem as described above on Raspberry PI, which works fine on UP2 Grove IoT Development Kit.  I wanted to check whether the pull request mentioning Raspberry PI (#928) would address the problem that I am currently see.

BinduRao2018, 2018-09-19:
Thanks for the clarification @Chuckduey.
Adding more details after my initial investigation:
I am using the latest board, 3 Model B+ (Revision a020d3), however mraa log the board information as  'Raspberry Pi Model B Rev 1' of type 5. Is this correct?
I had a look at the implementation - mraa_raspberry_pi() and revision a020d3 is not checked to get the Model. Is something missing?

Chuckduey, 2018-09-19:
It looks like they may have slipped another Rev of Pi 3+  I also just found an a020d3 Rev as well.  I will add that to the code, you should see it in a couple of days.   Thanks!

Chuckduey, 2018-09-20:
Thanks for your look into this.  Something broke, and I am not sure of this code right now.  I will cancel my pull request until I can make sure it is working correctly.  When I pulled down the code from the repo everything is there, but it does not recognize the Pi 3 B+.  The code that it says is the same is working fin.  I think it is better to pull it rather than put bad code out ther.

Best regards

Chuck

BinduRao2018, 2018-09-21:
Thanks for letting me know. I will wait for the update then.

Chuckduey, 2018-09-22:
I put my pull request back up.  It looks like my code is detecting the Raspberry Pi 3 + with the Revision a020d3 correctly.  With Pull Request #928 it should detect the Raspberry Pi correctly and not the Default Raspberry Pi  Model B Rev 1.  The old Pi B only has 26 pins and had a slightly different GPIO scheme.  Try out the new code and see if it detects the Pi correctly as a Raspberry Pi 3 Model B+.  This might fix you problem.  

Best regards

Chuck

BinduRao2018, 2018-09-24:
Hi @Chuckduey,
I confirm that the fix available on #928 detects Raspberry PI 3+ board as mentioned by you. I did run one of the my test program to drive an LED after initialization, which works fine. However, I haven't verified all the interfaces(connectors) available via Grove PI.  
Could you let me know when this fix will be merged to master and officially available as a release please?

BinduRao2018, 2018-10-08:
Is there any update on this please?

Propanu, 2018-10-09:
Hi @BinduRao2018, this was merged to master with d3207769e7a9df9f4afe10d502964f51ef0dca76.

BinduRao2018, 2018-10-11:
I am now using d320776 on master branch. Looks good to me on UP2 & Raspberry PI.

alext-mkrs, 2018-10-13:
Looks like solved - closing

closed 2018-10-13 by alext-mkrs