eclipse-mraa/mraa #931: Info required: Use of Grove - Ultrasonic Ranger without ISR

BinduRao2018, 2018-09-25:
Hello,
I would like to test Grove - Ultrasonic Ranger using mraa on UP2 Grove Kit. As I see ISR is not yet mapped with mraa, could you let me know the way to test this sensor please?

BinduRao2018, 2018-09-25:
Hi @Propanu ,
Thanks for your quick response. 
I am trying GrovePI+ shield on multiple platforms. So, I wouldn't be able to try an option to connect to the UP2 board directly. 
I went through hcsr04 driver implementation, which I would like to explore. I have one question on this:  hcsr04.c expects trigger pin and echoPin - 2 pin configuration, however with the Ultrasonic ranger I will be using only one pin that acts as trigger first and then read the values. Would this work ok?

Propanu, 2018-09-25:
Maybe not then, at least not out of the box. You might have to modify the hcsr04 library slightly and recompile it to use a single pin. The way the ultrasonic ranger works is the same though, as I remember at some point we were discussing merging the 2 drivers into 1 (hcsr04 and ultrasonic). @malikabhi05 has worked with these 2 libraries in the past and might be able to help.

BinduRao2018, 2018-09-26:
I tried by modifying the hcsr04 implementation by using the same pin for trigger and then echo. However, it doesn't seem to work. gpio doesn't seem to get the correct value (1 in this case) to calculate the distance. @malikabhi05 could you provide some inputs on this please?

BinduRao2018, 2018-10-11:
Any input on this please? Is there any plan to include ISR support using mraa in the near future?