eclipse-mraa/mraa #932: Info required: No of I2C devices that can be accessed simultaneously on Grove PI+ platform using mraa

BinduRao2018, 2018-09-26:
I have connected  > 1 device/sensor  on Grove PI+ - I2C ports (Port I2C-2, I2C-1, I2C-3). I would like to read from/write to these devices simultaneously. However, I see that only one I2C_BUS i.e Bus 0 is initialized for GrovePI, which can be used at a time to map the device slave address to the bus.  Does that mean, If connected via I2C, then only one device can be accessed at any point in time using mraa?

Propanu, 2018-10-09:
Hi @BinduRao2018, you can definitely use multiple devices on the I2C bus. The biggest limiting factor is the total capacitance of the bus, as every device you add will make the signals a bit weaker. There are definitely exceptions though as some devices, especially ones that are not 100% compliant with the I2C protocol may not allow you to attach additional devices to the same bus.

tingleby, 2018-10-10:
Also worth noting, if the i2c device(sensor) has the same address on more than one, that can cause a problem too.

BinduRao2018, 2018-10-11:
Thanks for the information.

alext-mkrs, 2018-10-13:
Looks like answered - closing

closed 2018-10-13 by alext-mkrs