eclipse-mraa/mraa #933: Information required: pin mapping of a platform to GrovePI

BinduRao2018, 2018-10-17:
Hi,
I am in the process of using mraa implementation to support GrovePI on Artik board. I have the implementation using mraa that works fine on UP2 board and Raspberry PI. Idea is to use the same implementation and update mraa to support Artik, as the support is not available.
I went thr' thr 'raspberry_pi.c' and 'up2.c' respectively to understand the pin mapping. However, it is still not clear to me on how a Grove Port (say D3), maps to a pin on Raspberry PI/UP2 board. 
i.e if i connect a LED on D3, I use mraa_gpio_init(515 + 3) to initialize. How does this map to the actual pin on UP2/Raspberry PI board?
Can I get some details please? 

Thanks,
Bindu

BinduRao2018, 2018-10-24:
I have now gone through the mraa impln for UP2 and the pin mappings are as described in https://wiki.up-community.org/Pinout_UP2. Can anybody, let me know on the mapping between UP2 pin number and sysfs (Linux GPIO no) please?

Thanks,
Bindu

Propanu, 2018-10-31:
> How does this map to the actual pin on UP2/Raspberry PI board?

GrovePi is an I2C subplatform, meaning you can use it on any MRAA supported platform that has an I2C bus available you can connect the shield to. To use pins on a subplatform you index them by 512 just like in your example (thus D0 is 512, D3 is 515 and so on). There is no direct mapping between platform pins and subplatform pins, but rather an implementation that enables communication between the platform and subplatform. In other words, MRAA talks to the GrovePi firmware running on the atmega328 via I2C to read values, write values and change pin modes.

Thus for the Artik board, you don't have to worry about the GrovePi at all. If you write a board definition for the Artik that exposes one of the i2c buses, you can plug the shield into that bus and initialize it with the generic `mraa_add_subplatform()` function.