eclipse-mraa/mraa #934: Publish mraa 2.0.0 to Maven Central

jakekarnes42, 2018-10-20:
It appears that MRAA has released version 2.0.0, but Maven Central only has up to version 1.9.0. Could you please publish the latest version (2.0.0) to Maven Central as well?

Thanks for the great project!

Propanu, 2018-10-25:
Hi @jakekarnes42 I managed to publish MRAA 2.0 for Android Things, but we'll need to update our build system to get the regular java bindings up on Maven. @tingleby is this something you can help with?

jakekarnes42, 2018-11-05:
Thanks @Propanu.

Any updates on this @tingleby? Your help would be greatly appreciated.

Propanu, 2018-11-06:
Yep, it'll end up on Maven Central too: https://oss.sonatype.org/#nexus-search;quick~mraa

jakekarnes42, 2018-11-10:
Confirmed that MRAA 2.0.0 is available on Maven Central here: https://mvnrepository.com/artifact/io.mraa/mraa/2.0.0

Thanks so much!

closed 2018-11-10 by jakekarnes42