eclipse-mraa/mraa #935: Kernel/config changes required on UP2 board to work with GrovePi

BinduRao2018, 2018-10-29:
I have tested GrovePi with the sensors on UP2 using mraa. When I try with GrovePi available on github (DexterInd/GrovePi), i get i2c errors, which indicates the problem in i2c configuration. Using mraa,  _mraa_add_subplatform(MRAA_GROVEPI, "0")_ is called during initialization to set the configuration.  Is it possible to get the changes required to configure GrovePI please to use with (DexterInd/GrovePi).

Note: I have updated the SMBUS to use the correct i2c device (/dev/i2c-5 based on i2cdetect).

Propanu, 2018-10-31:
Hi @BinduRao2018 there's a known issue with the GrovePI+ shield where you have to reset the board after using `i2cdetect` on the bus the shield is connected to, or you will get bus arbitration errors. Is this the error you are seeing?

BinduRao2018, 2018-10-31:
Hi @Propanu,

It turned out to be a permission issue. When I executed my program as root, there is no problem and the test works fine! This query could be closed.

Thanks ,
Bindu

Propanu, 2018-10-31:
Thanks for the update!

closed 2018-10-31 by Propanu