eclipse-mraa/mraa #936: Errors while installing on Raspberry Pi

gitlucky96, 2018-11-03:
I am trying to install mraa on a Raspberry Pi using the command line
`sudo add-apt-repository ppa:mraa/mraa`

However, I am getting an error saying that there is no version for Raspbian, even though the Raspberry Pi seems to be supported (acc. to Documentation).

`Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 89, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/stretch`

What am I doing wrong? Can someone help me here?

alext-mkrs, 2018-11-04:
As you can see in the [README](https://github.com/intel-iot-devkit/mraa/blob/master/README.md), the `add-apt-repository` is used on Ubuntu, but you're running Raspbian and quick googling shows that's not supposed to work there.

You can add that PPA manually though, see e.g. [here](https://raspberrypi.stackexchange.com/questions/44622/how-to-add-ppa-entries-manually-on-raspberry-pi#46933) for a recipe.

gitlucky96, 2018-11-04:
Thanks! I was able to install it after installing dirmngr (with sudo apt-get install dirmngr).
However, my c++ files don't seem to be able to use the library. Is there a way I can check if it is installed properly? And can I just include the mraa library with a #include directive in c++?

Thanks for your answers. I appreciate it!

alext-mkrs, 2018-11-15:
No worries.

If you could elaborate on how exactly

> my c++ files don't seem to be able to use the library

I should be able to advise. Generally there are examples packaged, so you could try those first.

kmh4ah, 2019-03-08:
> As you can see in the [README](https://github.com/intel-iot-devkit/mraa/blob/master/README.md), the `add-apt-repository` is used on Ubuntu, but you're running Raspbian and quick googling shows [...]

I still cannot solve it using the methods provided from the link. 
When I try to install 
sudo apt-add-repository ppa:pj-assis/ppa 

The error receives as below...

Traceback (most recent call last):
  File "/usr/bin/apt-add-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 89, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/stretch

cschaf, 2019-06-28:
Is there a solution found?

g-vidal, 2019-12-09:
Not exactly the answer   to the question but if you are in a hurry and have an online translator (for my french pages)  you can check the raspberryPi binary images I am trying to mainain for our project on meteorology and climate sciences, I deliver them with  debian and compilation of mraa and upm among other goodies. 
http://mediaserv.climatetmeteo.fr/images/RaspBerry/
And if you have the sufficient skills you can follow my blog and compile it by yourself  #digitalTransformation :-) 
https://blog.tremplin.ens-lyon.fr/GerardVidal/RaspberryPi-2_3BuildImage.html