eclipse-mraa/mraa #937: Error for gpio read on UP squared

yoann-solana, 2018-11-06:
Hi,
I am trying to read value on a pin with interruption but the read function is returning a negative value and not 0 or 1 as expected.

# Setup
* The board is an Intel Up squared running Ubuntu 16.04 with kernel 
```
$ uname -a  
Linux upsquared-UP-APL01 4.10.0-42-generic #5000~upboard11-Ubuntu SMP Mon Jun 25 15:58:51 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
I have followed all the [procedure](https://wiki.up-community.org/Ubuntu) to setup the UP.

To install MRAA, I followed the procedure from the [Readme](https://github.com/intel-iot-devkit/mraa#installing-on-ubuntu). The version is 2.0 (returned by mraa_get_version()).

# The code
The code (at the bottom) read the value of a pin on an interruption.During the reading I got an error. I chose the pin MRAA no.13 according to the [mapping](https://github.com/intel-iot-devkit/mraa/blob/master/docs/up2.md). I also tried with gpio 15, 16, 18 but still with errors. Also I am executing the code as a super user with sudo -E -S since it is a ROS node.

# Error
In the Interrupt Service Routine, I am using the mraa_gpio_read() function but it returns a negative value (error) and not 0 or 1. 
The log show an error when the mraa is initialized: 
*imraa: Failed to open lock file*
```
journalctl -t libmraa
-- Logs begin at mar. 2018-11-06 13:10:12 CET, end at mar. 2018-11-06 13:29:07 CET. --
nov. 06 13:29:07 th09001-PC libmraa[2538]: libmraa version v2.0.0 initialised by user 'root' with EUID 0
nov. 06 13:29:07 th09001-PC libmraa[2538]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1
nov. 06 13:29:07 th09001-PC libmraa[2538]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0
nov. 06 13:29:07 th09001-PC libmraa[2538]: up2: kernel pinctrl driver available
nov. 06 13:29:07 th09001-PC libmraa[2538]: imraa: Failed to open lock file
nov. 06 13:29:07 th09001-PC libmraa[2538]: gpio: support for chardev interface is activated
nov. 06 13:29:07 th09001-PC libmraa[2538]: libmraa initialised for platform 'UP2' of type 16
```
and this is what I get from the same log when there is an interruption on the gpio
```
nov. 06 13:34:40 th09001-PC libmraa[2538]: [GPIOD_INTERFACE]: ioctl() error Device or resource
nov. 06 13:34:40 th09001-PC libmraa[2538]: gpiod: ioctl() fail
nov. 06 13:34:40 th09001-PC libmraa[2538]: [GPIOD_INTERFACE]: error getting gpio line handle
nov. 06 13:34:40 th09001-PC libmraa[2538]: [GPIOD_INTERFACE]: ioctl() error Device or resource
nov. 06 13:34:40 th09001-PC libmraa[2538]: gpiod: ioctl() fail

```
Do you have information about the "fialed to open lock file"? 

Do you have an idea on how to solve this issue?

Thanks in advance.

```
#include <mraa.h>
#include <cassert>
#include <iostream>
#include <unistd.h>

/** Device context */
typedef struct _dev_context {
    mraa_gpio_context gpio;
    volatile long int position;
    int gpio_val_;
} *dev_context;

dev_context dev_init(int pin_a);
void dev_close(dev_context dev);
void dev_isr(void *ctx);

mraa_gpio_context d_pin;
dev_context enc;

void inputEnter(const std::string& str);
void checkRoot(void);

int main()
{
    // Check if running as root
    checkRoot();

    int gpio_pin = 13;

    enc = dev_init(gpio_pin);
    if( !enc)
    {
      std::cout<<"dev could not be initialized\n";
      return -1;
    }

    // Loop forever printing the counter value every second
    for (;;) {
        std::cout << "counter value " << enc->position << std::endl;
        sleep(1);
    }

    return MRAA_SUCCESS;
}

//------------------------------------------------------------------------------

void inputEnter(const std::string& str)
{
    std::cerr << str << "\nPress Enter to continue..." << std::endl;
    std::cin.get();
}

void checkRoot(void)
{
    int euid = geteuid();
    if (euid) {
        inputEnter("This project uses Mraa I/O operations that require\n"
            "'root' privileges, but you are running as non - root user.\n"
            "Passwordless keys(RSA key pairs) are recommended \n"
            "to securely connect to your target with root privileges. \n"
            "See the project's Readme for more info.\n\n");
    }
    return;
}

dev_context dev_init(int pin)
{
    dev_context dev = (dev_context)malloc(sizeof(struct _dev_context));

    if (!dev)
        return NULL;

    dev->gpio = NULL;
    dev->gpio_val_ = 0;
    dev->position = 0;

    // Make sure MRAA is initialized
    int mraa_rv;
    if ((mraa_rv = mraa_init()) != MRAA_SUCCESS)
    {
        printf("%s: mraa_init() failed (%d).\n", __FUNCTION__, mraa_rv);
        dev_close(dev);
        return NULL;
    }

    // Check platform and version
    if (mraa_get_platform_type() != MRAA_UP2) {
        printf("This example is meant for the UP Squared board.\n");
        return NULL;
    }
    const char* v = mraa_get_version();
    if (1) {
        printf("MRAA version = %s, should be >= 1.9.0!\n", v);
//        return NULL;
    }

    // MRAA contexts...
    if ( !(dev->gpio = mraa_gpio_init(pin)) )
    {
        printf("%s: mraa_gpio_init(pin_a) failed\n", __FUNCTION__);
        dev_close(dev);
        return NULL;
    }

    mraa_gpio_dir(dev->gpio, MRAA_GPIO_IN);

    // Setup the ISR
    // MRAA_GPIO_EDGE_BOTH for better resolution,
    // Could be not supported universally
    if (mraa_gpio_isr(dev->gpio, MRAA_GPIO_EDGE_BOTH, &dev_isr, dev))
    {
        printf("%s: mraa_gpio_isr() failed\n", __FUNCTION__);
        dev_close(dev);
        return NULL;
    }

    return dev;
}

void dev_close(dev_context dev)
{
    assert (dev != NULL);

    if (dev->gpio)
    {
        mraa_gpio_isr_exit(dev->gpio);
        mraa_gpio_close(dev->gpio);
    }

    free(dev);
}

void dev_isr(void *ctx)
{
    assert (ctx != NULL);
    dev_context dev = (dev_context)ctx;

    dev->gpio_val_ = mraa_gpio_read(dev->gpio);
    printf("gpio_val = %d\n", dev->gpio_val_);
    if(dev->gpio_val_ < 0 )
    {
      printf("Fatal error when reading value on gpio\n");
      return;
    }

    dev->gpio_val_ = dev->gpio_val_;
}

```

tingleby, 2018-11-07:
Perhaps checking the address of the ctx may help, using either printf or a debugger.

Propanu, 2018-11-08:
> Do you have information about the "fialed to open lock file"?

This is caused by `imraa`, the MRAA daemon used to flash and configure subplatforms, and is totally unrelated to this.

The issue is on the UP2 you cannot read a pin that has an ISR attached to it, although the callback as you might have noticed still runs. It was relatively easy to figure out why this wasn't known, as most MRAA ISR examples are cleverly crafted to print and not use arguments cross threads. So this is either an MRAA bug in the chardev implementation leading to `ioctl()` being used incorrectly or a firmware limitation on the UP2.

To help with this, the sysfs implementation should work as expected but you have to compile MRAA from source and install it after changing [this](https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/up2.c#L121) to 0. Something quicker I can think of trying would be to try and register separate ISRs for rising/falling edge modes, as all modes are supported by chardev too. If this does work there would be no need for the gpio ctx in the isr function anymore.

meshaun9, 2018-11-28:
I'm having the same issue currently but i'm using the python3 bindings for libmraa to control servo motors with the adafruit PCA9685 but initially I just was testing PWM on the actual UP2 board before I communicate through I2C to the Adafruit board so I ran the following via python3 interpreter:

`import mraa
x = mraa.Pwm(33)
`
I get the error that it couldnt' initialize and just like @yoann-solana i'm getting the same error when doing the journalctl command where the file is locked. So I was wondering does compiling libmraa from the source with the proposed change from @Propanu work? I plan to test this but want to know if someone has gone down this path and seen success or failure?

Propanu, 2018-11-28:
Hi @meshaun9, PWM should work out of the box and is not related to this issue. Sounds like you're not using the right kernel driver for the low speed I/O header. Can you please post the output of `journalctl` when you run your code?

alext-mkrs, 2018-12-25:
This sounds like resolved to me - closing. Feel free to comment if you need further help.

alext-mkrs, 2019-01-06:
@yoann-solana, ping?

@Propanu, do you want to pursue this one as per your earlier comment in case this is no longer actual for @yoann-solana?

alext-mkrs, 2019-05-05:
Okay, closing for no activity - feel free to reopen if still relevant.

closed 2019-05-05 by alext-mkrs

AndreasAZiegler, 2020-07-21:
I'm facing this problem as well. I use GPIO 431 as interrupt and also read its state. This only works if I use mraa compile with ``b->chardev_capable = 0;```  in ```src/x86/up2.c```. I will make a PR.