eclipse-mraa/mraa #939: Building mraa made de10-nano unknown platform

bozkurthan, 2018-11-13:
Hello everyone,

I'm trying to build mraa on DE10-Nano board. Actually, Mraa comes with DE10-Nano image, but I want to change Mraa for custom works such as adding more Uart interface to Arduino header. And I found document for DE10-Nano on Intel [https://software.intel.com/en-us/node/731128](url). Also, the link refers to building Mraa site. [https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md](url). So, firstly, I cloned the Mraa to the board. And then, changed some lines in "/src/arm/de_nano_soc.c".

```
    strncpy(b->pins[pos].name, "D0/RX", 8);
    b->pins[pos].capabilities = (mraa_pincapabilities_t){ 1, 0, 0, 0, 0, 0, 0, 1 };
    //b->pins[pos].gpio.pinmap = 0;
    //b->pins[pos].gpio.mux_total = 0;
    pos++;

    strncpy(b->pins[pos].name, "D1/TX", 8);
    b->pins[pos].capabilities = (mraa_pincapabilities_t){ 1, 0, 0, 0, 0, 0, 0, 1 };
    //b->pins[pos].gpio.pinmap = 0;
    //b->pins[pos].gpio.mux_total = 0;
    pos++;

    **strncpy(b->pins[pos].name, "D2/RX", 8);
    b->pins[pos].capabilities = (mraa_pincapabilities_t){ 1, 0, 0, 0, 0, 0, 0, 1 };
   // b->pins[pos].gpio.pinmap = 203;
    //b->pins[pos].gpio.mux_total = 0;
    pos++;

    strncpy(b->pins[pos].name, "D3/TX", 8);
    b->pins[pos].capabilities = (mraa_pincapabilities_t){ 1, 0, 0, 0, 0, 0, 0, 1 };
   // b->pins[pos].gpio.pinmap = 204;
   //b->pins[pos].gpio.mux_total = 0;
    pos++;**
```

Also added the Uart devices with following lines.

```
    // Arduino header UART (default)
    **b->uart_dev_count = 2;
    b->def_uart_dev = 1;
    b->uart_dev[0].device_path = "/dev/ttyS1";
    b->uart_dev[1].device_path = "/dev/ttyS2";**
    //b->uart_dev[0].rx = 0;
    //b->uart_dev[0].tx = 0;
```

After these changes, I think it's time to building Mraa. And I followed these steps;

mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DBUILDSWIGNODE=OFF -DBUILD_SHARED_LIBS=ON ..
make 
make install

But, after all these steps, `mraa-gpio version` command is said that Mraa version 2.0.0.4 and running on **Unknown Platform**.

Also, I looked cmake logs and found that the Target arch is armv7. The more logs will add after, because I'm still waiting to clean microSD image.

bozkurthan, 2018-11-13:
Before the build;

```
root@de10-nano:/$ mraa-gpio version
Version v1.5.1 on DE10-Nano-SoC
```

After the build, but I tried cmake .. with no flags;

```
root@de10-nano:~/newer/mraa/build$ cmake ..
-- The C compiler identification is GNU 6.2.1
-- The CXX compiler identification is GNU 6.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test HAS_C__Wall
-- Performing Test HAS_C__Wall - Success
-- Performing Test HAS_C__Werror_main
-- Performing Test HAS_C__Werror_main - Success
-- Performing Test HAS_C__Wformat
-- Performing Test HAS_C__Wformat - Success
-- Performing Test HAS_C__Wmain
-- Performing Test HAS_C__Wmain - Success
-- Performing Test HAS_C__Wuninitialized
-- Performing Test HAS_C__Wuninitialized - Success
-- Performing Test HAS_C__Winit_self
-- Performing Test HAS_C__Winit_self - Success
-- Performing Test HAS_C__Werror_implicit
-- Performing Test HAS_C__Werror_implicit - Success
-- Performing Test HAS_C__Werror_missing_parameter_type
-- Performing Test HAS_C__Werror_missing_parameter_type - Success
-- Performing Test HAS_CXX__Wall
-- Performing Test HAS_CXX__Wall - Success
-- Performing Test HAS_CXX__Werror_main
-- Performing Test HAS_CXX__Werror_main - Success
-- Performing Test HAS_CXX__Wformat
-- Performing Test HAS_CXX__Wformat - Success
-- Performing Test HAS_CXX__Wmain
-- Performing Test HAS_CXX__Wmain - Success
-- Performing Test HAS_CXX__Wuninitialized
-- Performing Test HAS_CXX__Wuninitialized - Success
-- Performing Test HAS_CXX__Winit_self
-- Performing Test HAS_CXX__Winit_self - Success
-- Performing Test HAS_CXX__Wnon_virtual_dtor
-- Performing Test HAS_CXX__Wnon_virtual_dtor - Success
-- Performing Test HAS_CXX__Woverloaded_virtual
-- Performing Test HAS_CXX__Woverloaded_virtual - Success
-- Performing Test HAS_CXX__Wreorder
-- Performing Test HAS_CXX__Wreorder - Success
-- INFO - libmraa Version v2.0.0-4-gd320776
-- INFO - cmake Version 3.6.1
-- INFO - Target arch is armv7
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/libpython2.7.so (found suitable exact version "2.7.12") 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named distutils.sysconfig
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Found PythonLibs: /usr/lib/libpython3.5m.so (found suitable exact version "3.5.2") 
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
INFO - Adding ARM platforms
-- Found SWIG: /usr/bin/swig (found version "3.0.8") 
-- PYTHON2 attempting to build!
-- PYTHON3 attempting to build!
-- INFO - NODE_ROOT_DIR is /usr/include/node
-- Found Nodejs: /usr/bin/node (found version "4.6.1") 
-- Found v8: /usr/include/node/v8.h (found version "4.5.103.37")
-- INFO - swig Version 3.0.8
-- INFO - CXX compiler Version GNU 6.2.1
-- INFO - Using V8 version > 3 so requiring C++11 compiler
-- INFO - install NODE modules to /usr/include/node/lib/node_modules/mraa/
-- Could NOT find GTest (missing:  GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- Install Google Test to enable additional unit testing
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/newer/mraa/build
```

That's important to know, it's detect only armv7 platform;

-- INFO - libmraa Version v2.0.0-4-gd320776
-- INFO - cmake Version 3.6.1
-- INFO - Target arch is armv7

But as far as I know, it should detect DE10-Nano-SoC .

tingleby, 2018-11-13:
Very interesting, :/
It would first appear that the upstream mraa does not detect this platform correctly.
Could you post the contents of `/proc/cpuinfo` please?
Also you could examine the syslog to see if there is anything of interest there

bozkurthan, 2018-11-14:
Thanks for replying.

cat /proc/cpuinfo gave me;

```
root@de10-nano:/proc$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1594.16
Features	: half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

processor	: 1
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1594.16
Features	: half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

Hardware	: Altera SOCFPGA
Revision	: 0000
Serial		: 0000000000000000
```

Also I checked `mraa/build/CMakeFiles/CMakeError.log` directory for error log which is gave me the following content;

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/root/mraa/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_dc31d/fast"
/usr/bin/make -f CMakeFiles/cmTC_dc31d.dir/build.make CMakeFiles/cmTC_dc31d.dir/build
make[1]: Entering directory '/home/root/mraa/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_dc31d.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_dc31d.dir/CheckSymbolExists.c.o   -c /home/root/mraa/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_dc31d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_dc31d.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_dc31d.dir/CheckSymbolExists.c.o  -o cmTC_dc31d -rdynamic 
CMakeFiles/cmTC_dc31d.dir/CheckSymbolExists.c.o:CheckSymbolExists.c:function main: error: undefined reference to 'pthread_create'
CMakeFiles/cmTC_dc31d.dir/CheckSymbolExists.c.o:CheckSymbolExists.c:function main: error: undefined reference to 'pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_dc31d.dir/build.make:98: cmTC_dc31d] Error 1
make[1]: Leaving directory '/home/root/mraa/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:126: cmTC_dc31d/fast] Error 2

File /home/root/mraa/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/root/mraa/build/CMakeFiles/CMakeTmp 

[... 80 lines cut ...]
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(mraa.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(gpio.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(gpio_chardev.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(i2c.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(pwm.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(spi.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(aio.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(uart.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(led.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(grovepi.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(jsonplatform.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(arm.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(96boards.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(raspberry_pi.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
[... cut ...]
/usr/lib/gcc/arm-angstrom-linux-gnueabi/6.2.1/../../../../arm-angstrom-linux-gnueabi/bin/ld: error: ../../libmraa.a(de_nano_soc.c.o): requires unsupported dynamic reloc R_ARM_MOVW_ABS_NC; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[2]: *** [src/python/python2/CMakeFiles/_python2-mraa.dir/build.make:132: src/python/python2/_mraa.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:232: src/python/python2/CMakeFiles/_python2-mraa.dir/all] Error 2
make: *** [Makefile:139: all] Error 2
```

If I use **-DCMAKE_C_FLAGS=-fPIC** flag, there is no problem for building..

tingleby, 2018-11-15:
@Propanu looks like you had some experience with this platform. Does the HW detection look to include this?

alext-mkrs, 2018-11-15:
I have this board, though I haven't used it for quite a while - but IIRC mraa was just fine there, including the compilation... Apparently the [identification ](https://github.com/intel-iot-devkit/mraa/blob/master/src/arm/arm.c#L76)in `cpuinfo` has changed since the support was added originally.

And apparently we don't support the "force" CMake switch for anything but x86 (I haven't noticed that before - does anyone know what's the idea behind this?).

@bozkurthan as you're anyway messing up with the source :) just change the line in `arm.c` I referred to above to check for "Altera SOCFPGA" instead of "DE0/DE10-Nano-SoC" as a quick fix. Feel free to submit a PR for that as well :)

I also don't fully get your last comment, you first state

> After all these processes, I cloned mraa again and after used cmake -DCMAKE_C_FLAGS=-fPIC -DBUILD_SHARED_LIBS=OFF -DBUILDSWIGPYTHON=OFF .. command following error occurred;

so you already have `-fPIC`, but then you state

> If I use -DCMAKE_C_FLAGS=-fPIC flag, there is no problem for building..

so you add the same flag again? How is it actually?

Propanu, 2018-11-16:
I was about to suggest forcing the platform so good catch there @alext-mkrs. I don't remember why it's not there, maybe just never added. @Mani-Sadhasivam would probably find that useful for 96boards too.
@bozkurthan, as far as I remember `CMAKE_C_FLAGS` and `CMAKE_CXX_FLAGS` are separate. perhaps that's what caused the confusion, in your log the build fails on the CXX target for the Python bindings.
Adding the `Altera SOCFPGA` string to the platform check as suggested by Alex would be the easiest way to get the platform recognized again, but let me ask, are you using a custom bitstream for the Cyclone V? If the memory addresses for the gpio registers on the FPGA change the board file needs to be updated too.

bozkurthan, 2018-11-19:
@alext-mkrs thanks for helping.

> so you add the same flag again? How is it actually?

Yes, I added same flag. But I used only -fPIC flag when I built second one. Actually, I think It's not about only flags. There was a no problem before the multiple-build. I think I messed up the Libraries and kind of system resources when I built source more than once. However, I'll try the force today and I'll be keep in touch. When I dig the problem further and decide for sure that there is a definitely problem, I'll submit PR. :)

> are you using a custom bitstream for the Cyclone V? 

@Propanu  You got me, but I have 2 answers for it.

**First answer,** Yes, I built for custom FPGA design and I'm planning to use on custom FPGA design. But, the fact is that MRAA runs well in custom design. 

For custom design, I followed the Intel tutorial in this link [https://github.com/intel-iot-devkit/terasic-de10-nano-kit/tree/master/tutorials](url) and I prepared to 4 switches, 8 leds, 1 button and 1 reset button. The key word is a DTBO (Device Tree Binary Overlay). When you load custom design, you must use DTBO for MRAA. Intel released DTBO in tutorials. 

After the load bitstream, the whole FPGA system changed as expected. But MRAA accepts custom designs with the` mraa-gpioraw ` command or `mraa_gpio_init_raw(int pin)` code segment. Due to MRAA uses first design of Board, the normal commands and codes are not running on custom design.

**Second answer**, is no. I tried in normal design of DE10-Nano. I'm using RBF file without preload. I mean I didn't change design. If I force the board to start with custom FPGA design in Uboot it's starts with custom design. But, I thought as you, it's a problem with custom design and decided to start with normal FPGA design. But, there was no change.

And lastly, I thought I messed up the board system file and maybe load and use clean OS image would better. After the clean image,  I hoped everything is OK. MRAA was running well. But after build, bang, again same problem occurred.

bozkurthan, 2018-12-27:
@alext-mkrs Actually, not exactly that. MRAA is running wells both of Custom FPGA design and default FPGA design. But, when i build MRAA from github by source it's getting fail. The problem is 
`Hardware	: Altera SOCFPGA` where the DE10-Nano board is defined `DE0/DE10-Nano-SoC` in mraa library.

alext-mkrs, 2019-01-06:
Thanks, okay, so we're seemingly now down to just one things of the several we've discussed along the way. Good :)

Okay, so this still sounds like newer kernel/OS build than the one used for mraa development originally. 

Could you please do two things:
1. Post the output of `uname -a` command from the board
1. Try to update the platform check code as I suggested [here](https://github.com/intel-iot-devkit/mraa/issues/939#issuecomment-439187242) and see if that works. If it does, we'll just need to update the library and a PR from you would be more than welcome - or at least this confirmation. Alternatively I could dig up and update my board, but that would take some noticeable time, so if you're still actively using it I think that'd be faster.

alext-mkrs, 2019-01-06:
Ok, never mind the second thing - I now see your PR #944 (catching up with GH notifications, sorry). I'll comment in the PR itself.

bozkurthan, 2019-01-17:
Hello, thanks for all your suggests. Actually, I'm newbie on github. This is my first Issue topic and PR. So, I don't want to do something wrong.

Here the output of `uname -a` :

```
root@de10-nano:~/Desktop$ uname -a
Linux de10-nano 4.1.33-ltsi-altera #1 SMP Thu Mar 30 10:37:56 PDT 2017 armv7l GNU/Linux
```

I'll read your PR comment and try to resolve.

bozkurthan, 2019-03-27:
PR #944 fixed this issue. I'll try later but for a now MRAA detect my board.

closed 2019-03-27 by bozkurthan