eclipse-mraa/mraa #940: Memory leaks in mraa_spi_write_buf and mraa_spi_write_buf_word

HelloHungryImDad, 2018-11-23:
Memory leaks in mraa_spi_write_buf (mraa/src/spi/spi.c:434) and mraa_spi_write_buf_word (mraa/src/spi/spi.c:451)  due to recv buffer being allocated and only freed on transfer error. On successful transfer, the recv buffer is returned by the functions but not freed from heap.

Given that these functions are wrappers for the equivalent transfer functions (lines 378 and 406 respectively) but discarding the received buffer, I would suggest making these functions void type and freeing the received buffer whether or not the transfer was successful. 

If the received buffer is required, one should just use the transfer functions.

tingleby, 2018-12-03:
Please feel free to create a PR if you feel improvements should be made :)

The API docs for these calls do mention the caller should free the returned buffer:
https://github.com/intel-iot-devkit/mraa/blob/master/api/mraa/spi.h#L133

HelloHungryImDad, 2018-12-04:
Thanks for pointing that out, I'd missed that one!

I'll close this and raise a PR as I think this might ineccesarily catch a few people out. I'd seen a few issues around where people had noticed memory leaks in the js wrappers and I wonder if this gotcha was inherited.

Amazing project by the way. I wish I'd discovered this library sooner!

closed 2018-12-04 by HelloHungryImDad