eclipse-mraa/mraa #943: The number of UART on minnowboard turbot is wrong.

Windows10010, 2018-12-26:
While I was testing the mraa framework written in java, I found that there was only one on-board uart interface that is available in the program as three uarts **_(one for FTDI serial,two for low-speed expansion)_** are supposed to be available according to the port layout.

[image]

When the code is like this:
`Uart uart = new Uart(0);`
program runs without fail and all the functionalities work normally.

But when I wanted to try another port, say
`Uart uart = new Uart(1);`
The program throws an Exception of `Error initializing uart 1` and exited with en error code.
I also enumerated the value from 0 to 10 as an argument to the constructor of Uart class, but none of them worked. 

Also I checked out the result of the method `mraa.getUartCount();` of only **1** instead of **3**.
Hardwares using UART to communicate should not be connected to the debug serial interface for production purposes. Is this a bug? If so, I hope it can be fixed in few weeks.

alext-mkrs, 2019-01-06:
@Windows10010 mraa doesn't directly use those pins to produce UART "traffic" - it rather attaches to OS-level devices that are exposed by the Linux device driver(s). Turbot is indeed currently only configured to have one port in the UART definitions in mraa (not sure why, @arfoll might remember) - but you could try to use the name-based init to use other UARTs: `new Uart('device name, e.g. /dev/ttyS1')`.

Please give it a try and let us know if that works.

Windows10010, 2019-02-08:
Oh God, it doesn't work at all even if I pass `0` as the argument like the last time........

tingleby, 2019-02-08:
So regrettably I haven't had any experience with this board. But you can find out what serial devices are available on the board by running `ls /dev/ttyS*`

Judging by https://elinux.org/Minnowboard:MinnowMax
UART 1 your looking for might be accessible via - `new Uart("/dev/ttyS4")`

arfoll, 2019-02-14:
The problem on minnowboard is you can change those pins using the BIOS. I think i just picked whatever came default and left it a that. Uarts are super easy to init with the device node rather than pin numbers. Could just be easier to init the uarts with matching IDs in sysfs - I guess improvements are welcome :)

alext-mkrs, 2019-05-05:
So I think we can close this one with the recommendation to use device node names as listed above. Closing and feel free to post if you need further help.

closed 2019-05-05 by alext-mkrs